adafruit/circuitpython #7608: WeAct Studio pico boards wrong VID .?

Neradoc, 2023-02-18:
The weact studio boards use the Adafruit VID, with a PID from Raspberry Pi, assigned to the YD RP2040

This one: https://github.com/raspberrypi/usb-pid/commit/690a77c1e3061db7aa20557bd422ba0b707a190c

https://github.com/adafruit/circuitpython/blob/508f2015b68042c25256a8f57dd37e5aec2c48c5/ports/raspberrypi/boards/vcc_gnd_yd_rp2040/mpconfigboard.mk#L1-L2
https://github.com/adafruit/circuitpython/blob/508f2015b68042c25256a8f57dd37e5aec2c48c5/ports/raspberrypi/boards/weact_studio_pico/mpconfigboard.mk#L1-L2
https://github.com/adafruit/circuitpython/blob/508f2015b68042c25256a8f57dd37e5aec2c48c5/ports/raspberrypi/boards/weact_studio_pico_16mb/mpconfigboard.mk#L1-L2

tannewt, 2023-02-21:
This looks related: https://github.com/WeActStudio/WeActStudio.RP2040CoreBoard/issues/2

Maybe the quickest way to fix is to request a pid.codes PID.

dhalbert, 2023-04-11:
@fabaff Are you willing to do a PR to pid.codes for WeAct PID's? I don't think they are going to do anything.