adafruit/circuitpython #7609: ColorConverter.convert() returns different values from same input

FoamyGuy, 2023-02-18:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.2 on 2023-02-14; Adafruit Feather ESP32-S2 TFT with ESP32S2
Board ID:adafruit_feather_esp32s2_tft

Adafruit CircuitPython 8.0.2 on 2023-02-14; Adafruit Feather RP2040 with rp2040
Board ID:adafruit_feather_rp2040
```

### Code/REPL

```python
import time
from displayio import ColorConverter, Colorspace

cc_888 = ColorConverter(input_colorspace=Colorspace.RGB888, dither=False)

results_histo = {}
for i in range(1100):
    val = cc_888.convert(0xff0000)
    if val not in results_histo:
        print(f"{i}: {hex(val)} | {val} | {bin(val)}")
        results_histo[val] = 1
    else:
        results_histo[val] += 1

        # enabling or disabling the print influences the 'failure' rate
        print(f"{i}: {hex(val)} | {val} | {bin(val)}")

print("------------")
print(results_histo)
```

### Behavior

With the code above I would expect to receive the same result value back for every call to `ColorConverter.convert()` because we always input the exact same value to it's argument. But in reality sometimes we are getting a different value returned from `convert()` sometimes. Here is a sample of the output after 1100 iterations: 
```
{248: 1061, 63488: 39}
```
So the vast majority of the time it is the same, but occasionally different. The Feather ESP32-S2 TFT seems to have an overall higher rate of differing values. With print enabled I was seeing closer to 60/40 split between the differing values. The Feather RP2040 seems like it's somewhat less likely, more often being around 50ish out of 1100 different.

### Additional information

@gamblor21 and I discussed on discord a bit and he found a potential cause and solution. Will make a PR with that after some more testing.

gamblor21, 2023-02-18:
PR will be submitted soon but the problem was in `shared-bindings/displayio/ColorConverter.c` in `displayio_colorconverter_obj_convert(...)`

```C
    _displayio_colorspace_t colorspace;
    colorspace.depth = 16;
```

The colorspace object is never initialized and later in the code `colorspace.reverse_bytes_in_word` is referenced. If it is incorrectly set to non-zero the function fails.

FoamyGuy, 2023-02-20:
resolved by #7611

closed 2023-02-20 by FoamyGuy