adafruit/circuitpython #7612: UART write does not work on QT Py SAMD21 with CircuitPython 8.x

alanmitchell, 2023-02-19:
### CircuitPython version

```python
Tested with CP 8.0.2 and CP 20230218-12d3640.

Code works properly with CP 7.3.3.
```

### Code/REPL

```python
import time
import board
import busio

e5_uart = busio.UART(
    board.TX, board.RX,
    baudrate=9600,
    timeout=0.01
)

while True:
    out_bytes = bytes('From QT Py\n\r', 'utf-8')
    e5_uart.write(out_bytes)
    in_bytes = e5_uart.read(10)
    if in_bytes:
        print(in_bytes)

    time.sleep(1)
```

### Behavior

I have the QT Py Tx and Rx pins connected to a USB-TTL serial converter connected to my Linux PC.  For CP 8.x versions, no characters are received on the PC.  When I send characters from the PC to the USB-TTL converter, the QT Py does properly receive them, as indicated by characters being printed on the QT Py Serial console.

The above code works properly with CP 7.3.3.

dhalbert, 2023-02-19:
Reproduced. Thanks -- I am taking a look.

dhalbert, 2023-02-23:
Fixed by #7616.

PR #7616 (2023-02-22): Fix pad assignments on atmel-samd UART
- Fixes #7612.

- SERCOM USART TXPO computation was not correct on SAMD21 for TX pins using pad 2, due to changes introduced in #6434. SAMD21 and SAMx5x are different: SAMx5x does not allow pad 2 for TX. In addition, checking for allowed pins based on their pads was not as rigorous as it should have been. Code is somewhat restructured.
- Signature for `busio.UART()` was missing several arguments and did not account for `None` values.
- Did more pin validation in `shared-bindings` `busio.UART()`. Ensure pins are distinct, and that both TX and RX are not No [... cut]
closed 2023-02-23 by dhalbert