adafruit/circuitpython #7614: MatrixPortal M4 - ImportError: no module named 'platform' - CircuitPython 8.0.2

csteamengine, 2023-02-19:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.2 on 2023-02-14; Adafruit Matrix Portal M4 with samd51j19
```

### Code/REPL

```python
import time
import board
import terminalio
from adafruit_matrixportal.matrixportal import MatrixPortal

# --- Display setup ---
matrixportal = MatrixPortal(status_neopixel=board.NEOPIXEL, debug=True)

# Create a new label with the color and text selected
matrixportal.add_text(
    text_font=terminalio.FONT,
    text_position=(0, (matrixportal.graphics.display.height // 2) - 1),
    scrolling=True,
)

SCROLL_DELAY = 0.03

contents = [
    { 'text': 'THIS IS RED',  'color': '#cf2727'},
    { 'text': 'THIS IS BLUE', 'color': '#0846e4'},
]

while True:
    for content in contents:
        matrixportal.set_text(content['text'])

        # Set the text color
        matrixportal.set_text_color(content['color'])

        # Scroll it
        matrixportal.scroll_text(SCROLL_DELAY)
```

### Behavior

Traceback (most recent call last):
  File "code.py", line 4, in <module>
  File "/lib/adafruit_matrixportal/matrixportal.py", line 33, in <module>
  File "/lib/adafruit_matrixportal/network.py", line 31, in <module>
  File "adafruit_portalbase/network.py", line 28, in <module>
  File "/lib/adafruit_io/__init__.py", line 21, in <module>
  File "/lib/adafruit_io/client.py", line 24, in <module>
ImportError: no module named 'platform'

### Description

This is the simplest example code from the MatrixPortal Circuitpython library found here: 
https://docs.circuitpython.org/projects/matrixportal/en/latest/

I've copied all the suggested libraries into the `lib/` directory on the CIRCUITPY drive, but can't get past this error.

csteamengine, 2023-02-19:
Installed 7.3.3 as well and it has the same error: 
`Adafruit CircuitPython 7.3.3 on 2022-08-29; Adafruit Matrix Portal M4 with samd51j19`
```
Traceback (most recent call last):
  File "code.py", line 4, in <module>
  File "/lib/adafruit_matrixportal/matrixportal.py", line 33, in <module>
  File "/lib/adafruit_matrixportal/network.py", line 31, in <module>
  File "adafruit_portalbase/network.py", line 28, in <module>
  File "/lib/adafruit_io/__init__.py", line 21, in <module>
  File "/lib/adafruit_io/client.py", line 24, in <module>
ImportError: no module named 'platform'
```

Neradoc, 2023-02-19:
How did you install the adafruit_io library ? It looks like you took the wrong files.
The library should be copied from the bundle: https://circuitpython.org/libraries

csteamengine, 2023-02-19:
Copied them over from `adafruit-circuitpython-bundle-py-20230218` initially, and just retried with `adafruit-circuitpython-bundle-8.x-mpy-20230219 but wasn't working. 

Then I decided to completely delete the `lib` directory and go back through 1 by 1 and adding them from the most recent library folder and it worked!

csteamengine, 2023-02-19:
Not sure which library was wrong, but getting rid of all of them seemed to fix it

Neradoc, 2023-02-19:
Looks like you got this: https://github.com/adafruit/Adafruit_IO_Python (which is not in the bundle)
Instead of this: https://github.com/adafruit/Adafruit_CircuitPython_AdafruitIO
Glad you got it working now.

closed 2023-02-19 by Neradoc