adafruit/circuitpython #7620: cannot start access point mode for Raspberry Pi Pico W. the error is: RuntimeError: Wifi is in station mode.

didakus, 2023-02-21:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.2 on 2023-02-14; Raspberry Pi Pico W with rp2040
Board ID:raspberry_pi_pico_w
```

### Code/REPL

```python
import os
import wifi

ssid = os.getenv('CIRCUITPY_WIFI_SSID')
password = os.getenv('CIRCUITPY_WIFI_PASSWORD')

print("Activating Access Point SSID...", ssid)
wifi.radio.start_ap(ssid, password)
```

### Behavior

code.py output:
Activating Access Point SSID... ReS_AP01
Traceback (most recent call last):
  File "code.py", line 11, in <module>
RuntimeError: Wifi is in station mode.

Code done running

### Description

I can not start an AP in my raspberry pi pico w. I tried to add the next lines, before the start_ap method:

tried 1:
wifi.radio.stop_station()
wifi.radio.enabled = False
wifi.radio.enabled = True
wifi.radio.start_ap(ssid, password)

tried 2:
wifi.radio.enabled = False
wifi.radio.stop_station()
wifi.radio.enabled = True
wifi.radio.start_ap(ssid, password)

and with multiple secuences i always receive the same error.

### Additional information

i was searching for an start_ap example in the adafruit-circuitpython-bundle-examples and in the documentation and i did not find anything about it. It could be possible to add some code?

anecdata, 2023-02-21:
There are currently limitations in the Pico W implementation due primarily to external submodules. This will hopefully improve in the future. But for now, here's a summary of the limitations:
https://github.com/adafruit/circuitpython/pull/7101#pullrequestreview-1188988867

Basically, a device can either be a Station or an Access point, and since the example above starts web workflow and therefore becomes a Station, it precludes the device from becoming an AP.

didakus, 2023-02-21:
Thank you so much @anecdata . It Works for me your explanation and the code shared. I deleted the lines of wifi configuration in the settings.toml and changed my code to this lines:

if wifi.radio.ipv4_address_ap is None:
   print("Activating Access Point SSID...")
   wifi.radio.start_ap("Cookie", "franemma")
else:
    print("Access Point SSID already active...")

This scheme is neccesary to avoid an error when AP is already started. And when a i need a change in AP parameters i code a microcontroller.reset()

closed 2023-02-21 by didakus

anecdata, 2023-02-21:
There are some edge cases where the IP address may be `None` or "0.0.0.0" while the connection is being made or possibly while the connection is being broken. Milliseconds typically, but you might encounter it.