adafruit/circuitpython #7630: Support ulaw audio in .wav

dholth, 2023-02-22:
ulaw is a very simple audio codec that originates in telephony. Here is one open implementation. Typically would use 64kbps at 8 bits per sample, 8khz sampling. https://github.com/dpwe/dpwelib/blob/master/ulaw.c

It would be a nice way to halve audio bitrates without using the memory or compute of mp3, and could include both the encoder and decoder.

tannewt, 2023-02-22:
Does Audacity support it? How would one create a file for it?

dholth, 2023-02-22:
In audacity you "export as wav" with "encoding: u-law". The https://en.wikipedia.org/wiki/Au_file_format is maybe a more "classic" way to store this data, available as "Other uncompressed files" with AU header, encoding: u-law. (a-law is very similar but I think u-law is slightly more common / better). Consider also setting the sample rate to 8khz in Audacity's Tracks -> Resample