adafruit/circuitpython #7634: Support SPI word lengths besides 8 or 9 bits

cafeclimber, 2023-02-22:
Working with a device that uses a SPI bus with 6-bit command words. When trying to configure the bus for this via:

```python
...
    spi = busio.SPI(board.SCK, MOSI=board.MOSI)
    spi.configure(bits=6)
...
```
I get:
```python
ValueError: Invalid number of bits
```

([This is the device](https://cdn.macom.com/datasheets/MAPS-010164.pdf) I'm trying to control, if that's helpful)

dhalbert, 2023-02-23:
We don't support any microcontrollers that have sub-8-bit SPI peripherals. In general, 8-bit is standard, a few support 9-bit or 16-bit.

Despite the fact that the datasheet says the device "is compatible with the SPI protocol", it is so only in a loose sense.

If you use 8-bit SPI and have your setting bits in the lower 6 bits, does that work? Maybe it will accept 8 clocks and just discard the top two bits.

You could also simulate 6-bit SPI by bitbanging the data and clock lines yourself. Even our `bitbangio.SPI()` does not support sub 8-bit SPI.

tannewt, 2023-02-23:
What board are you using? PIO on the RP2040 could be used too.

cafeclimber, 2023-02-24:
@tannewt I'm using a Teensy 4.1.

@dhalbert thanks for this! I had considered that I might have to do some bit banging, but to be honest I rarely do much work with micros. When I looked at the source for busio, I saw it was doing what looked like a simple check for the bits argument and wondered if it might be as easy as adjusting that check. 

I'll get a chance to look at the device tomorrow and was going to try an 8-bit word like you mentioned and hope it discarded the unnecessary bits. If not, I've got the other avenues you mentioned. Thanks for the help!

cafeclimber, 2023-02-25:
@dhalbert ended up trying your suggestion of just ignoring the top two bits of an 8 bit word, and it worked a treat!

Thanks again.

closed 2023-02-25 by cafeclimber

dhalbert, 2023-02-25:
Great! I wish the datasheet had been clearer. So often they say only just the bare minimum, without helpful use cases.