adafruit/circuitpython #7635: Metro ESP32-S2 <Beta>Hard Crash on CircuitPython Internet Test

electromechpro, 2023-02-22:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.2 on 2023-02-14; Adafruit Metro ESP32S2 with ESP32S2
Board ID:adafruit_metro_esp32s2
UID:C7FD1AA13E86
```

### Code/REPL

```python
# SPDX-FileCopyrightText: 2020 Brent Rubell for Adafruit Industries
#
# SPDX-License-Identifier: MIT

import ipaddress
import ssl
import wifi
import socketpool
import adafruit_requests

# URLs to fetch from
TEXT_URL = "http://wifitest.adafruit.com/testwifi/index.html"
JSON_QUOTES_URL = "https://www.adafruit.com/api/quotes.php"
JSON_STARS_URL = "https://api.github.com/repos/adafruit/circuitpython"

# Get wifi details and more from a secrets.py file
try:
    from secrets import secrets
except ImportError:
    print("WiFi secrets are kept in secrets.py, please add them there!")
    raise

print("ESP32-S2 WebClient Test")

print("My MAC addr:", [hex(i) for i in wifi.radio.mac_address])

print("Available WiFi networks:")
for network in wifi.radio.start_scanning_networks():
    print("\t%s\t\tRSSI: %d\tChannel: %d" % (str(network.ssid, "utf-8"),
            network.rssi, network.channel))
wifi.radio.stop_scanning_networks()

print("Connecting to %s"%secrets["ssid"])
wifi.radio.connect(secrets["ssid"], secrets["password"])
print("Connected to %s!"%secrets["ssid"])
print("My IP address is", wifi.radio.ipv4_address)

ipv4 = ipaddress.ip_address("8.8.4.4")
print("Ping google.com: %f ms" % (wifi.radio.ping(ipv4)*1000))

pool = socketpool.SocketPool(wifi.radio)
requests = adafruit_requests.Session(pool, ssl.create_default_context())

print("Fetching text from", TEXT_URL)
response = requests.get(TEXT_URL)
print("-" * 40)
print(response.text)
print("-" * 40)

print("Fetching json from", JSON_QUOTES_URL)
response = requests.get(JSON_QUOTES_URL)
print("-" * 40)
print(response.json())
print("-" * 40)

print()

print("Fetching and parsing json from", JSON_STARS_URL)
response = requests.get(JSON_STARS_URL)
print("-" * 40)
print("CircuitPython GitHub Stars", response.json()["stargazers_count"])
print("-" * 40)

print("done")
```

### Behavior

Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Crash into the HardFault_Handler.
Please file an issue with the contents of your CIRCUITPY drive at 
https://github.com/adafruit/circuitpython/issues

Press any key to enter the REPL. Use CTRL-D to reload.

### Additional information

Was unable to access SPIFFS when using in Arduino mode (github signal generator with wifi controlls.)  That project had to have the Arduino code changed for the S2 version (no high speed ledc, etc), but the web page would not display (AP).

The normal blink code works in circuitpython.  I use AP mode for my Arduino projects, but there is not a lot of support for that yet.

dhalbert, 2023-02-23:
I could not reproduce this on a Metro ESP32-S2 with 8.0.2 and the latest version of `adafruit_requests` (installed with `circup`).  You might want to try erasing the flash completely and then reinstalling the UF2 bootloader: https://learn.adafruit.com/adafruit-metro-esp32-s2/install-uf2-bootloader

electromechpro, 2023-02-27:
It turns out there is something wrong on that board.  Repeated erasing/installing the UF2 bootloader gave the same results.  Thank you for looking at the problem.   I am glad you were able to get the example to work on your end.  That lead me to replace the board. The test program is now working.

dhalbert, 2023-02-27:
Thanks for retesting and verifying it works elsewhere.

closed 2023-02-27 by dhalbert