adafruit/circuitpython #7637: web workflow API endpoint to return free space on device

vladak, 2023-02-23:
https://github.com/adafruit/circup code has a TODO to check free space on a device before installing a module. To support this for web workflow (https://github.com/adafruit/circup/issues/156) as well, the value available needs to be available via the [REST API](https://docs.circuitpython.org/en/latest/docs/workflows.html#file-rest-api).

tannewt, 2023-02-23:
This would use [`f_getfree()`](http://elm-chan.org/fsw/ff/doc/getfree.html) internally. We may want to add it to the BLE workflow API too.

Neradoc, 2023-03-12:
Free space (and total space) would be useful to show in the file list page too (or at least on cp.org via /code ?)
I don't know if we want the info to be in /cp/version.json or a separate url.

makermelissa, 2023-03-13:
From the code editor standpoint, I would think another (similar) URL would be better because the version stuff usually doesn't change much once the board is booted and it can be a bunch to parse through. Maybe something like /cp/diskinfo.json and then other filesystem stuff could go there unless it adds a lot of overhead and makes the CP too big.

PR #7746 (2023-03-21): Diskinfo api for web workflow
resolves #7637 

URL is `/cp/diskinfo.json` Response sizes are in bytes.

Example of response:
```
{"free": 212992, "block_size": 512, "total": 963072}
```
Maybe we don't really need `block_size` in the results? I can remove it if not. I added it initially in order to test how it's looked up, which is needed to find the correct multiplier to get to bytes units.

Tested successfully with Feather ESP32-S2 TFT.

Thank you to: isacben, Neradoc, and anecdata who all helped me in getting this far on implementing this API. Also Scott for the pointer to t_getfree(), the F [... cut]