adafruit/circuitpython #7638: socketpool, ssl: rationalize return types & throwing exceptions

jepler, 2023-02-23:
While looking into #7606 @dhalbert noted that there were irregularities in our socket-like APIs that may have contributed to the bug existing in the first place.

We should consider rationalizing the code so that:
 * the common-hal APIs consistently return either positive counts or negative errno values, in signed types (mp_int_t)
 * the shared-bindings have the responsibility of transforming negative errno values into OSError exceptions
 * noting, at least in comments, common-hal APIs that are not permitted to assume the VM is running (no alloc, no exception)

gneverov, 2023-02-23:
Without the `common_hal_` prefix the functions can sometimes collide with functions in 'shared-bindings' which already typically don't have a prefix.

tannewt, 2023-02-23:
Yup! I usually add a `_` prefix which I use for statics ala Python's private designation.

dhalbert, 2023-02-23:
Maybe we could use a `base_` prefix or similar for the functions that can be called from C outside the VM. Maybe you can think of a better prefix. I too usually drop the `common_hal_` for routines not meant to be called via a Python API.