adafruit/circuitpython #7650: Can't use SPI and I2C at same time?

djairjr, 2023-02-25:
### CircuitPython version

```python
Adafruit CircuitPython 8.1.0-alpha.2-14-gc2c7b9345 on 2023-02-22; Seeeduino XIAO RP2040 with rp2040
Board ID:seeeduino_xiao_rp2040
```

### Code/REPL

```python
# I am posting the code for Gifio test, but the same problem happens with any use of TFT Display and I2C call at same time...

import board
import busio
import gifio
import displayio
import time
import adafruit_ili9341

dc=board.D5
rst=board.D4
cs=board.D2 #My display does not have it...

displayio.release_displays()
spi = busio.SPI(clock=board.SCK, MOSI=board.MOSI, MISO=board.MISO)
i2c = busio.I2C( board.SCL, board.SDA, frequency=200_000)

display_bus = displayio.FourWire(spi, command = dc, chip_select=cs, reset=rst)

display = adafruit_ili9341.ILI9341(display_bus, width=320, height=240)
display.root_group.hidden = True # HIDE REPL
display.auto_refresh = False
splash = displayio.Group()
display.show(splash)

odg = gifio.OnDiskGif('/images/circlescv.gif')
odg.next_frame() # Load the first frame

#face = displayio.TileGrid(odg.bitmap, pixel_shader=displayio.ColorConverter(input_colorspace=displayio.Colorspace.RGB565))
face = displayio.TileGrid(odg.bitmap, pixel_shader=displayio.ColorConverter(input_colorspace=displayio.Colorspace.RGB565))
splash.append(face)

# Wait forever
while True:
    odg.next_frame()
    display.refresh()
    #time.sleep(0.1)
```

### Behavior

``` python
Traceback (a última chamada mais recente):
  Arquivo "<stdin>", linha 16, em <module>
ValueError: SCL em uso
```

Meaning SCL in use...

### Description

Can't use SPI Bus together with I2C Bus. 
In Seeed Xiao RP2040 
I2C pins are: Board.D4 = Board.SDA and Board.D5 = Board.SCL
SPI Pins are:  Board.D10 = Board.MOSI, Board.D9 = Board. MISO, Board.D8 = Board.SCK and Board.D7 = Board. CS (but any pin can do. The expansion board uses Board.D2 as CS Pin)

### Additional information

I'm working on a controller for Escape Room artifacts with multiple peripherals using the I2C bus. They all work perfectly. When I try to add a TFT display to the SPI bus, I get this message. I haven't performed the test yet with other devices on the SPI bus to see if that and repeats.

deshipu, 2023-02-25:
you forgot to call `displayio.release_displays()`

deshipu, 2023-02-25:
You can't use the same `board.D5` pin as both D/C and SCL at the same time.

djairjr, 2023-02-25:
Sorry man! When I post it, I've discovered my stupidity. Hahahaha

djairjr, 2023-02-25:
I will solve it and post if it works.

closed 2023-02-27 by tannewt