adafruit/circuitpython #7653: Setting PWM duty_cycle causes a blocking delay proportional to PWM frequency

CedarGroveStudios, 2023-02-27:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.2 on 2023-02-14; Adafruit Feather M4 Express with samd51j19
```

### Code/REPL

```python
# SPDX-FileCopyrightText: 2018 Kattni Rembor for Adafruit Industries
#
# SPDX-License-Identifier: MIT

"""CircuitPython Essentials: PWM with Fixed Frequency example."""
import time
import board
import pwmio

# LED setup for most CircuitPython boards:
led = pwmio.PWMOut(board.LED, frequency=5000, duty_cycle=0)

while True:
    t0 = time.monotonic()
    for i in range(100):
        # PWM LED up and down
        if i < 50:
            led.duty_cycle = int(i * 2 * 65535 / 100)  # Up
        else:
            led.duty_cycle = 65535 - int((i - 50) * 2 * 65535 / 100)  # Down
        #time.sleep(0.01)
    print(f"average duty_cycle set time: {time.monotonic() - t0:0.3f} sec")
```

### Behavior

100 iterations with a PWM frequency of 5000 Hz:
```Python
average duty_cycle set time: 0.020 sec
```

100 iterations with a PWM frequency to 50 Hz:
```Python
average duty_cycle set time: 2.004 sec
```

### Description

The delay is approximately equal to the period of the PWM frequency and introduces noticeable impact at lower frequencies such as those used for DC motor control.

It appears that the SAMD51 timer used for PWM is _not_ being operated in it's double-buffered CC (Compare Channel) register mode, but is waiting to load the CC register when the timer count is at it's ZERO value rather than loading the CC buffer register and returning to the calling module.  (reference: Microchip SAM D5x/E5x Family Data Sheet , DS60001507E, pp 1719-1721)

If possible, the desired operation would be to load the new PWM duty_cycle value and immediately return to the calling code, allowing the SAMD51's timer hardware to self-synchronize rather than cause a blocking delay condition.

tannewt, 2023-02-28:
This must be related to SYNCBUSY:

https://github.com/adafruit/circuitpython/blob/8a8579a9b9bc3bd2200426d7294dd16f8bd72c6c/ports/atmel-samd/common-hal/pwmio/PWMOut.c#L293-L335

I think we do want the value to be copied on reload but that shouldn't prevent us from writing the buffered value multiple times in the interim. We just changed this on rp2: #7299

CedarGroveStudios, 2023-02-28:
Also suspect that the delay leads to DC motor "rattling" when establishing a new motor speed at low PWM frequencies since the two PWM control pins aren't changed simultaneously.

CedarGroveStudios, 2023-03-07:
> Also suspect that the delay leads to DC motor "rattling" when establishing a new motor speed at low PWM frequencies since the two PWM control pins aren't changed simultaneously.

Confirmed. At low PWM frequencies in slow decay mode, the delay causes a DC motor to noticeably chatter when switching between velocity zero and coasting (velocity = None) and back. Don't have a setup to test stepping motors, but would expect a similar issue.

timchinowsky, 2024-05-19:
I have been doing some benchmarking and the slowness of PWM setting immediately popped out.  Looks like it is a known issue which still need to be fixed?  And is there any background on if/how the SAMD PWM peripheral is different from e.g. the RP2040's, which requires setting it to be handled differently?

tannewt, 2024-05-20:
> I have been doing some benchmarking and the slowness of PWM setting immediately popped out. Looks like it is a known issue which still need to be fixed? And is there any background on if/how the [...]

I don't think it needs to be different. It just hasn't been updated. See https://github.com/adafruit/circuitpython/issues/7653#issuecomment-1447322516

timchinowsky, 2024-05-22:
Wondering why is double-buffering disabled here?
https://github.com/adafruit/circuitpython/blob/2854dbe37f52dc9ebf48035dfc82bba83c8f9cac/ports/atmel-samd/common-hal/pwmio/PWMOut.c#L278-L286

and why is it necessary to wait for sync?
https://github.com/adafruit/circuitpython/blob/2854dbe37f52dc9ebf48035dfc82bba83c8f9cac/ports/atmel-samd/common-hal/pwmio/PWMOut.c#L275-L276
Thanks!

tannewt, 2024-05-22:
> and why is it necessary to wait for sync?

I don't think it is necessary. It was just a choice we made during the original implementation.

> Wondering why is double-buffering disabled here?

No idea. Its been years since this code was added.

timchinowsky, 2024-05-22:
Ok!  I'm going to be looking at this and #7747