adafruit/circuitpython #7664: esp32-s3: crash if USB HID used too close to startup

jepler, 2023-03-01:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0-rc.1 on 2023-01-30; Adafruit QT Py ESP32S2 with ESP32S2
```

### Code/REPL

```python
Adafruit CircuitPython 8.0.0-rc.1 on 2023-01-30; Adafruit QT Py ESP32S2 with ESP32S2
import time
import board
import digitalio
import busio
from adafruit_hid.mouse import Mouse
from usb_hid import devices

import time
# time.sleep(3) # ??? needed delay at start to not crashy-crashy
SCALE = 1

mouse = Mouse(devices)

spi = board.SPI()
strobe = digitalio.DigitalInOut(board.RX)
strobe.switch_to_output(False)

data = bytearray(5)
if not spi.try_lock():
    raise SystemExit("could not lock SPI bus")

spi.configure(baudrate=100_000, polarity=1)

def delta(data):
    data = ~data
    if data & 0x80:
        return -(data & 0x7f)
    else:
        return data & 0x7f

x = y = 0

while True:
    strobe.value = True
    strobe.value = False
    spi.readinto(data, end=2)
    spi.readinto(data, start=2)
    lmb = bool(~data[1] & 0x40)
    rmb = bool(~data[1] & 0x80)
    dy = delta(data[2]) * SCALE
    dx = delta(data[3]) * SCALE
#    y += dy
#    x += dx
#    print(f"{'L' if lmb else ' '} {'R' if rmb else ' '} {dx:4} {dy:4} {x:5} {y:5}")
    mouse.report[0] = (
        mouse.MIDDLE_BUTTON if (lmb and rmb) else
        mouse.LEFT_BUTTON if lmb else
        mouse.RIGHT_BUTTON if rmb else
        0)

    if dx or dy:
        mouse.move(dx, dy)
    else:
        mouse._send_no_move() # pylint: disable=protected-access
```

### Behavior

Linux logs kernel errors, the device LED blinks red, and I have to enter safe mode to recover.

The problem goes away if I add the 'sleep' before initializing the mouse.

It may somehow be specific to the HW (SNES mouse clone), as a minimized version that has
````
mouse=Mouse(devices)
mouse._send_no_move()
```
as the last 2 lines (instead of talking to the SNES mouse hw) does not fail in the same way. 

### Description

```
Mar  1 10:10:23 localhost kernel: [2156930.784465] usb 3-2.1: new full-speed USB device number 5 using xhci_hcd
Mar  1 10:10:23 localhost kernel: [2156931.049421] usb 3-2.1: New USB device found, idVendor=239a, idProduct=8112, bcdDevice= 1.00
Mar  1 10:10:23 localhost kernel: [2156931.049427] usb 3-2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Mar  1 10:10:23 localhost kernel: [2156931.049429] usb 3-2.1: Product: Adafruit QT Py ESP32S2
Mar  1 10:10:23 localhost kernel: [2156931.049430] usb 3-2.1: Manufacturer: Adafruit
Mar  1 10:10:23 localhost kernel: [2156931.049432] usb 3-2.1: SerialNumber: 4D9FD80798C7
Mar  1 10:10:23 localhost kernel: [2156931.177499] cdc_acm 3-2.1:1.0: ttyACM1: USB ACM device
Mar  1 10:10:23 localhost kernel: [2156931.189501] usb-storage 3-2.1:1.2: USB Mass Storage device detected
Mar  1 10:10:23 localhost kernel: [2156931.189656] scsi host11: usb-storage 3-2.1:1.2
Mar  1 10:10:23 localhost kernel: [2156931.204838] input: Adafruit Adafruit QT Py ESP32S2 Keyboard as /devices/pci0000:00/0000:00:08.1/0000:0b:00.3/usb3/3-2/3-2.1/3-2.1:1.3/0003:239A:8112.0036/input/input160
Mar  1 10:10:23 localhost kernel: [2156931.264742] input: Adafruit Adafruit QT Py ESP32S2 Mouse as /devices/pci0000:00/0000:00:08.1/0000:0b:00.3/usb3/3-2/3-2.1/3-2.1:1.3/0003:239A:8112.0036/input/input161
Mar  1 10:10:23 localhost kernel: [2156931.264876] input: Adafruit Adafruit QT Py ESP32S2 Consumer Control as /devices/pci0000:00/0000:00:08.1/0000:0b:00.3/usb3/3-2/3-2.1/3-2.1:1.3/0003:239A:8112.0036/input/input162
Mar  1 10:10:23 localhost kernel: [2156931.264971] hid-generic 0003:239A:8112.0036: input,hidraw5: USB HID v1.11 Keyboard [Adafruit Adafruit QT Py ESP32S2] on usb-0000:0b:00.3-2.1/input3
Mar  1 10:10:24 localhost kernel: [2156932.202772] scsi 11:0:0:0: Direct-Access     Adafruit Adafruit QT Py E 1.0  PQ: 0 ANSI: 2
Mar  1 10:10:24 localhost kernel: [2156932.202974] sd 11:0:0:0: Attached scsi generic sg2 type 0
Mar  1 10:10:24 localhost kernel: [2156932.203304] sd 11:0:0:0: [sde] 4737 512-byte logical blocks: (2.43 MB/2.31 MiB)
Mar  1 10:10:24 localhost kernel: [2156932.204494] sd 11:0:0:0: [sde] Write Protect is off
Mar  1 10:10:24 localhost kernel: [2156932.204500] sd 11:0:0:0: [sde] Mode Sense: 03 00 00 00
Mar  1 10:10:24 localhost kernel: [2156932.205498] sd 11:0:0:0: [sde] No Caching mode page found
Mar  1 10:10:24 localhost kernel: [2156932.205501] sd 11:0:0:0: [sde] Assuming drive cache: write through
Mar  1 10:10:24 localhost kernel: [2156932.211685] usb 3-2.1: input irq status -75 received
Mar  1 10:10:25 localhost kernel: [2156933.116697] usb 3-2-port1: Cannot enable. Maybe the USB cable is bad?
Mar  1 10:10:26 localhost kernel: [2156933.988715] usb 3-2-port1: Cannot enable. Maybe the USB cable is bad?
Mar  1 10:10:27 localhost kernel: [2156934.864733] usb 3-2-port1: Cannot enable. Maybe the USB cable is bad?
Mar  1 10:10:28 localhost kernel: [2156935.736751] usb 3-2-port1: Cannot enable. Maybe the USB cable is bad?
```

solverit, 2023-09-09:
The same behavior for ESP32-S3 controllers, for example, M5 AtomS 3 Lite.

dhalbert, 2024-01-24:
I think this may be fixed by <https://github.com/adafruit/Adafruit_CircuitPython_HID/pull/118> and <https://github.com/adafruit/Adafruit_CircuitPython_HID/pull/122>, which now cause HID device lookups to be delayed until USB is ready. But I don't have the hardware that produces the problem to retest.

tannewt, 2024-01-31:
Please re-test this with a 9.0 beta release. If it is still an issue, then file a new issue referencing this one.

closed 2024-01-31 by tannewt