adafruit/circuitpython #7667: Split epaperdisplay from displayio

tannewt, 2023-03-01:
This will save space on boards that have an existing display and are unlikely to have an external epaper display (like pew pew m4).

tannewt, 2023-10-02:
We should go further with this as we have more boards with displays that aren't supported by `displayio.Display()`. (Dotclock displays use `FramebufferDisplay` for example. We should split `displayio.Display` out as well as `Epaperdisplay`.

tannewt, 2023-10-02:
`displayio.Display` support LCD and OLED displays that have driver ICs with onboard framebuffer RAM. We should rename it to reflect that.

Also, should we split out the different display busses as well?

makermelissa, 2023-10-03:
> Also, should we split out the different display busses as well?

That makes sense because paralleldisplay is already split out unless it would be more overhead.

deshipu, 2023-10-03:
Can we use `busdisplay` instead of `externalframebufferio`?

tannewt, 2023-10-04:
> Can we use `busdisplay` instead of `externalframebufferio`?

Ya, totally open to different module names.

commit e1df5981 (2023-10-25): Split displayio hardware support from core

These are moved:
* Display -> busdisplay.BusDisplay
* FourWire -> fourwire.FourWire
* EPaperDisplay -> epaperdisplay.EPaperDisplay
* I2CDisplay -> i2cdisplaybus.I2CDisplayBus

`paralleldisplay` is now `paralleldisplaybus` (and registered as
`paralleldisplay` too).

Bus related helpers are split out of display_core into bus_core.
It is in still displayio since it is a dependency of both
busdisplay and epaperdisplay.

Fixes #7667
closed 2023-10-25 by dhalbert