adafruit/circuitpython #7677: CircuitPython 8.0.3 wifi.radio.connect does not support 64 hex digit WPA2 PSK key strings

rich123, 2023-03-04:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.3 on 2023-02-23; Adafruit Feather ESP32S2 with ESP32S2
```

### Code/REPL

```python
>>> import wifi
>>> wifi.radio.connect(ssid="myssid",password="0123456789abcdef0123456789abcdef0123456789abcdef0123456789abcdef")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: password length must be 8-63
>>> wifi.radio.connect(ssid="myssid",password=b'\x01\x23\x45\x67\x89\xab\xcd\xef\x01\x23\x45\x67\x89\xab\xcd\xef\x01\x23\x45\x67\x89\xab\xcd\xef\x01\x23\x45\x67\x89\xab\xcd\xef')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ConnectionError: Unknown failure 15
>>> wifi.radio.connect(ssid="edbb364-f",password="weakpassword")
>>> wifi.radio.ipv4_address
10.0.0.36
>>>
```

### Behavior

First attempt, with a 64 hex digit PSK key string, returns a length error message that the password must be 8-63 characters.

Second attempt, with a b'' string, returns "Unknown failure 15"

Third attempt, after reconfiguring my AP to use a short, weak, password instead of a 64 hex digit key string, succeeds.

Note, same AP for all three tries.

### Description

First foray into Adafruit Feathers and CircuitPython.  Attempted to test connecting the Feather to my AP.  My first try was with a secrets.py with my 64 hex digit key inside.  I got the 8-63 length error.

After some Googling of both the CirPy docs for wifi.radio and for Python literals, I tried the b'' encoded variant (note, not my real hex key above).  It returns the "Unknown error 15" result.

After too many failed attempts, I decided to reconfigure my AP temporarily with a basic password instead of the hex key, and wifi.radio.connect() connected to the AP without issue.

Is there a way to pass WPA2 PSK hex key strings into wifi.radio.connect() so I can get the feather to connect using my normal key? 

If not, then please consider adding support for 64 digit hex keys.  I really don't want to reconfigure all of my other wireless devices because the CirPy does not appear to support an otherwise standard way of specifying a WPA2 PSK key.  Every other wireless device I have works with the 64 digit hex key.

dhalbert, 2023-03-06:
On Espressif, I think we just need to allow 64 characters, and probably should vet that the characters are all hex digits.
On RP2040 CYW43, it's not documented as far as I can tell whether this would work or not.

We would have the password be ASCII hex digits as opposed to raw binary.

dhalbert, 2023-04-14:
@rich123 Would you be interested in working on this?

rich123, 2023-04-14:
> @rich123 Would you be interested in working on this?

I can see if I can give it a go.  Will have to see what I need to install to actually build the code for the ESP chip first.

rich123, 2023-04-16:
The pull request I just pushed worked when built for my Adafruit ESP32-S2 Feather.  My 64-char WiFi password was accepted and the Feather authenticated with my AP.

closed 2023-04-20 by dhalbert