adafruit/circuitpython #7680: wifi.radio.start_ap - Cannot set AuthMode because it is not iterable

dotpointer, 2023-03-04:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.2 on 2023-02-14; Adafruit Feather ESP32S2 with ESP32S2
```

### Code/REPL

```python
import wifi
# 1 - give it an AuthMode
wifi.radio.start_ap(ssid='Wireless', password='wireless', authmode=wifi.AuthMode.WPA2)
# 2 - give it an AuthMode inside an iterable list to make it iterable as requested
wifi.radio.start_ap(ssid='Wireless', password='wireless', authmode=[wifi.AuthMode.WPA2])
```

### Behavior

# 1 - It refuses and requests the AuthMode to be iterable
TypeError: 'AuthMode' object is not iterable
# 2 - It is not satisified even when it is iterable
ValueError: Invalid authmode

### Description

The AP mode refuses to use AuthMode other than the default open unencrypted mode.

The problem might be a usage/syntax error, please tell me if so and how to fix it.

The errors are translated from Swedish.

dotpointer, 2023-03-04:
I found a solution, with WPA2 you also need to supply PSK:

`import`
`wifi.radio.start_ap(ssid='Wireless', password='wireless', authmode=[wifi.AuthMode.WPA2, wifi.AuthMode.PSK])`

dhalbert, 2023-03-04:
The documentation signature is wrong, it should say `authmodes`, so reopening.

dhalbert, 2023-03-04:
We should also maybe say that PSK is necessary.

PR #8105 (2023-06-23): improve start_ap() doc; make "authmode" use consistent internally
- Fixes #7680.

- Documentation corrections and improvement.
- Use `authmode` instead of `authmodes` internally. Originally I thought I would rename the `start_ap()` keyword argfrom `authmode` to `authmodes`. But the plural usage is not consistent with the naming used internally in Espressif and Pico W, and elsewhere. An "authmode" can be a combination like WPA/WPA2/PSK.

EDIT: forgot to say that I did test this on ESP32-S2.
closed 2023-06-23 by dhalbert