adafruit/circuitpython #7682: Labels on RGBMatrix flicker when writing/reading from SD card (sdcardio)

phantomsixthplayer, 2023-03-05:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.3 on 2023-02-23; Adafruit Grand Central M4 Express with samd51p20

adafruit-circuitpython-grandcentral_m4_express-en_GB-8.0.3.uf2
update-bootloader-grandcentral_m4-v3.15.0.uf2
```

### Code/REPL

```python
import displayio
import rgbmatrix
import board
import framebufferio
from adafruit_bitmap_font import bitmap_font
from adafruit_display_text.label import Label
import digitalio
import busio
import sdcardio
import storage
import adafruit_debouncer

displayio.release_displays()

matrix = rgbmatrix.RGBMatrix(
    width=64,
    height=32,
    bit_depth=4,
    rgb_pins=[board.D8, board.D9, board.D10, board.D11, board.D12, board.D13],
    addr_pins=[board.D4, board.D6, board.D3, board.D5],
    clock_pin=board.D1,
    latch_pin=board.D2,
    output_enable_pin=board.D0
)
display = framebufferio.FramebufferDisplay(matrix)

font = bitmap_font.load_font("/fonts/test.pcf")

mg = displayio.Group()
mg_a = Label(font, text="A.A.A.A", color=0x00FF00, x=15, y=6)
mg_h = Label(font, text="H.H.H.H", color=0x00FF00, x=14, y=17)
mg_b = Label(font, text="B.B.B.B", color=0x00FF00, x=15, y=28)
mg.append(mg_a)
mg.append(mg_h)
mg.append(mg_b)

class HiscoreController:
    def __init__(self):
        spi = busio.SPI(board.SD_SCK, board.SD_MOSI, board.SD_MISO)
        sdcard = sdcardio.SDCard(spi, board.SD_CS)
        vfs = storage.VfsFat(sdcard)
        storage.mount(vfs, "/")

    def hiscore(self, value=-1):
        if value >= 0:
            try:
                with open("/hiscore.txt", "w") as hiscore_file:
                    if value > 9999:
                        hiscore_file.write(str(value - 10000))
                    else:
                        hiscore_file.write(str(value))
            except OSError:
                pass
        else:
            if "hiscore.txt" in os.listdir("/"):
                try:
                    with open("/hiscore.txt", "r") as hiscore_file:
                        score = hiscore_file.read()
                        return score
                except OSError:
                    pass

class ButtonController:
    def __init__(self, btns_and_names):
        self.buttons = {}

        for name, btn in btns_and_names.items():
            button_pin = digitalio.DigitalInOut(btn["obj"])
            button_pin.switch_to_input(pull=digitalio.Pull.UP)
            self.buttons[name] = adafruit_debouncer.Debouncer(button_pin)

    def is_pressed(self, btn_name):
        return self.buttons[btn_name].fell

    def reset(self):
        for btn in self.buttons.values():
            btn.update()

    def update(self):
        for btn in self.buttons.values():
            btn.update()

hiscore_ctrl = HiscoreController()

buttons = ButtonController(
    btns_and_names = {
        "reset": {"num": 2, "obj": board.D27}
    }
)

def menu(screen_state):
    reset = False

    display.show(mg)

    buttons.reset()

    while screen_state == menu:
        buttons.update()

        if buttons.is_pressed("reset"):
            hiscore_ctrl.hiscore(value=0)
            reset = True

screens = {
    menu: menu
}

screen_state = menu

while True:
    current_screen = screens[screen_state]
    screen_state = current_screen(screen_state)
```

### Behavior

I switched to reading/writing to a file on an SD card because writing to CIRCUITPY would lock up other tasks. However, now when using the SD card whenever I try to write/read from the SD card the labels on the RGBMatrix flicker.
hiscore_ctrl.hiscore(value=0)
hiscore_ctrl.hiscore(value=game_score)

I used to write to the CIRCUITPY drive but it would cause other tasks to lock up.
import board
import digitalio
import storage

switch = digitalio.DigitalInOut(board.D36)
switch.direction = digitalio.Direction.INPUT
switch.pull = digitalio.Pull.UP

storage.remount("/", switch.value)

class HiscoreController:
    def __init__(self):
        pass

    def hiscore(self, value=-1):
        if value >= 0:
            hiscore_file = open("/hiscore/hiscore.txt", "w")
            if value > 9999:
                hiscore_file.write(str(value - 10000))
            else:
                hiscore_file.write(str(value))
            hiscore_file.close()
        else:
            hiscore_file = open("/hiscore/hiscore.txt", "r")
            score = hiscore_file.read()
            hiscore_file.close()
            return score

### Additional information

Example of flicker when pressing a button, easiest way to replicate.

https://user-images.githubusercontent.com/22541984/222948788-d7cb5321-ebe6-4a67-a06f-7c46f34e20a5.mp4

phantomsixthplayer, 2023-03-06:
Thought might be something to do with baudrate, but can't set it

in __init__
TypeError: extra keyword arguments given

class HiscoreController:
    def __init__(self):
        spi = busio.SPI(board.SCK, board.MOSI, board.MISO, baudrate=12000000)
        sdcard = sdcardio.SDCard(spi, board.CS, baudrate=12000000)
        vfs = storage.VfsFat(sdcard)
        storage.mount(vfs, "/")

dhalbert, 2023-03-06:
I don't think it's so likely this is an electrical problem, but rather it's a timing problem. Running an RGB Matrix requires real-time updates of the matrix, and missing an update can create flicker. It takes about 1/3 of the CPU to do it. It's done at interrupt level, so it should be capable of overriding most other things going on. but writing to an SD card may add uninterruptable things, or may cause contention that introduces latency and therefor flickering. We'll need to do some instrumentation to find out what is going on.