adafruit/circuitpython #7693: Feature request: Add Bluetooth support for Pico W

hex-developer, 2023-03-08:
Since the Raspberry Pi Foundation released the Pico SDK 1.5.0 with Bluetooth support, do you plan on releasing Circuitpython with Bluetooth support for the pico?

tannewt, 2023-03-08:
Not any time soon. I'm sure we'll do it at some point though. Unfortunately it uses BTStack and we haven't used it in CircuitPython before. (Only NimBLE and nordic SD.)

hex-developer, 2023-06-17:
Micropython just released support for Bluetooth on the Raspberry Pi Pico W. Here is the commit: [https://github.com/micropython/micropython/commit/856e08b1931b88271816a2f60648f6ff332235b2](https://github.com/micropython/micropython/commit/856e08b1931b88271816a2f60648f6ff332235b2). Maybe you can add Bluetooth support now.

dsz-dsz-k, 2023-06-20:
I can't wait for this to happen. It would enable KMK firmware to use bluetooth, thus enabling simple building of wireless keyboards using the Pico W...

tannewt, 2023-06-20:
We've put a lot of work into BLE on the nRF52840 and recommend folks using it. Supporting the Pico W isn't a high priority for us (but we're happy to guide and review if someone else wants to take it on.)

dsz-dsz-k, 2023-06-21:
Are there any guidelines / tasklists of what is required to do that?

WayneKeenan, 2023-06-21:
I've got experience in embedding Nimble in Micropython on the Micro:bit v1.  It's transferable skills to this use case, would it be possible for it to be sponsored for me todo it for CIrcuitPython?

tannewt, 2023-06-21:
> I've got experience in embedding Nimble in Micropython on the Micro:bit v1. It's transferable skills to this use case, would it be possible for it to be sponsored for me todo it for CIrcuitPython?

Maybe! Email pt@adafruit.com and he'll chat with you more about it.

tikimo, 2023-07-26:
Hi guys, a little bump on this open feature. Whats the status? Keyboard community is eagerly looking forward to this! :)

tannewt, 2023-07-26:
> Hi guys, a little bump on this open feature. Whats the status? Keyboard community is eagerly looking forward to this! :)

No one is working on it that I know of. The nRF52840 is a good existing option for BLE support.

adrianblakey, 2023-08-10:
Is there any way to get this implemented?

tikimo, 2023-08-10:
Even though the nRF## is a suitable workaround, it exhausts pins from the board and undermines the boards built-in capabilities. I hope someone can pick this up! ✨

montanag, 2023-08-29:
As someone writing applications for IOT devices, this in and of itself is a dealbreaker driving me to options like micropython (which doesn't have good tooling for BLE, but does in-fact support it). I imagine there anyone else writing applications for IOT devices utilizing bluetooth feels similarly.

dsz-dsz-k, 2023-09-02:
> Even though the nRF## is a suitable workaround, it exhausts pins from the board and undermines the boards built-in capabilities. I hope someone can pick this up! ✨

Hi
Could you explain what you mean by that? What is "it" in your sentence? Does the firmware prevent using pins on the nRF##?

tikimo, 2023-09-02:
> Hi Could you explain what you mean by that? What is "it" in your sentence? Does the firmware prevent using pins on the nRF##?

Using an external ble hardware uses pins of the main board, which in most of my projects is not possible, since those pins are already in use. Also, the external hardware takes up space which further complicates the setup. 

Furthermore, it makes no sense to work around the above constraints since the pico already has the hardware capability and even Micropython support.

dhalbert, 2023-09-03:
The nRF52840 boards run CircuitPython and have native BLE. They are not an add-on co-processor, so pins do not need to be used up for that.

JoshKamps, 2023-09-18:
just another comment in support of having it work on pico w. Got excited that circuit python had nice ble support but then realised it didn't work on pico w after installing :[

foxt, 2023-09-21:
@dhalbert The Pico W is not a nRF52840 board.

dhalbert, 2023-09-21:
There is no particular known stumbling block -- it's a matter of developer resources and priorities. BLE is also available on some Espressif chips: we have limited support, there is an issue of missing functionality in the underlying software. nRF52840 boards are available and have fine support.

hex-developer, 2023-09-21:
The circuitpython ([page](https://circuitpython.org/board/raspberry_pi_pico_w/)) says:
> Raspberry Pi Pico W brings WiFi + BLE (coming soon) wireless networking to the Pico platform while retaining complete pin compatibility with its older sibling.

If you don't plan on supporting BLE for the Pico W, then don't tell users to wait for it, as it will never happen.

bullwinkle3000, 2023-09-21:
Your only real response to my offer to help was to change the language from BLE "coming soon" to "not currently supported"?

You should close this issue, too.  It's clearly resolved.

dhalbert, 2023-09-21:
> Your only real response to my offer to help was to change the language from BLE "coming soon" to "not currently supported"?

The PR was to address https://github.com/adafruit/circuitpython/issues/7693#issuecomment-1729728523, not your offer to help.

> You should close this issue, too. It's clearly resolved.

No it's not, and please don't be snarky. Take us at our word. We are not giving an evasive answer. We are flat out on a bunch of other things for 9.0.0 right now.

If you'd like to help, we'd be happy to discuss this in [discord](https://adafru.it/discord).

The MicroPython implementation and especially the underlying library can be used as the starting point to re-implement `_bleio`. The license for the underlying library needs to be read carefully to make sure it is not too restrictive.

Another consideration is whether there is enough RAM and flash space available to have both wifi and BLE available at the same time. The co-processor is reloaded with a firmware blob on power-up, and that firmware has to live somewhere.

We ran into some issues on Espressif BLE due to restrictions about dynamic service creation in nimble. I don't know whether the same thing is an issue in btstack.

bullwinkle3000, 2023-09-21:
You're right, that was snark, and I apologize. 

I'm also frustrated and was piqued that your initial responses to my offer of help was to refer me to other boards and to remove coming soon from the Pico W page.

But I do apologize for the snark.  I'm sure you're busy as hell and snark isn't helpful or appreciated.

The firmware/memory thing could well be a problem just for my use case but I'll take a look.  I already have another firmware blob in the mix.

And thank you for the discord link.  I'll look into it there.

dhalbert, 2023-09-21:
@bullwinkle3000 Thanks for your response! The `#circuitpython-dev` channel in discord is where the devs hang out. `#help-with-circuitpython` is for end-user problems.

tannewt, 2023-09-21:
We're going to lock this issue but leave it open. If anyone wants to help add BLE support to Pico W, then please ping us on [Discord](https://adafru.it/discord). :point_up:

eightycc, 2023-09-25:
The CYW43 bluetooth firmware adds ~13KB to the existing ~225KB Wifi-only image. There is one new 6KB blob plus the Wifi + bluetooth blob grows by 7KB over the Wifi-only blob. This should partially answer @dhalbert 's concern about flash and RAM space. I'll have some numbers for btstack flash and RAM usage soon.

ItsTerm1n4l, 2023-09-26:
+1 
I think this would be a great feature (as well as one that I personally need :) as it would enable easy single-board bluetooth projects, especially with regards to [kmk](https://github.com/KMKfw/kmk_firmware).
keep up the good work!

eightycc, 2023-09-29:
With firmware CYW43 bluetooth firmware blobs, BTstack, and a skeleton implementation of _bleio in place flash grows from 85.7% to 88.14% and RAM from 40.14% to 44.35%. This includes both BLE and Wifi support.

I'm filling in functionality by cherry picking existing code wherever possible, otherwise writing new code. My method is adding to the port's _bleio one function at a time, testing, and then moving on.

No blocking issues at this time.

ItsTerm1n4l, 2023-10-17:
MicroPython has BLE support for the Pico and @dhalbert has just merged https://github.com/adafruit/circuitpython/pull/8481, Is this a start on BLE support for the Pico w?

tannewt, 2023-10-17:
> MicroPython has BLE support for the Pico and @dhalbert has just merged #8481, Is this a start on BLE support for the Pico w?

The MP merge is unrelated because we don't share the BLE subsystem with MP. eightycc's work to get BLE going is more promising.

markmcgookin, 2023-11-15:
@eightycc is there anything we can do as a community to help you with your work on this?

bullwinkle3000, 2023-11-15:
@eightycc  I'm hoping to use _bleio on the Pico W for wireless keyboard builds with KMK firmware (based on CP).  They've already got an early general BLE implementation for keyboards, including split-style keyboards, which are the kind that I build and sell.  Given the memory, pinouts, and flash available on the Pico W, it's a contender for wireless builds, even if that means a fat battery.  

I'd be happy to help, too, but my time is sorely limited.

I am curious if there are aspects of BLE on the Pico W that you see as particularly problematic?

eightycc, 2023-11-15:
@bullwinkle3000 Thank you for that info. Every little bit helps.

Problematic isn't the word I'd choose, I'll stick with *difficult*. There are a number of reasons for that. The BLE support is built on top of the Raspberry Pi SDK plus the CYW43-driver. Both are excellent projects but are a little thin on documentation for BLE. So, I've had to deep dive into the source for each. Infineon doesn't provide a public specification for the CYW43439's API, so I've had to make a lot of inferences by studying the CYW43-driver. Raspberry Pi SDK uses BTStack which is new to the CircuitPython project and to me, so again I'm hitting the source code. I'm modeling my work on the _bleio component of the nRF port, which is comprehensive and well-written but I'm having to code many of the HAL routines and all of the lower-level routines from scratch. Looking at the Micropython implementation is helpful, but the way it's structured goes against the grain of how CircuitPython does _bleio so not too much of it can simply be cribbed.

adriangalilea, 2023-11-15:
@eightycc
I'm hoping to use bluetooth as a remote macropad and to communicate communicate p2p with other pico's(or bt devices). Simple strings would work fine :)

ItsTerm1n4l, 2023-11-27:
@eightycc Im hoping to use bluetooth on the Pi pico W for a wireless keyboard with KMK.
I didn't realise there was not support for it until after I had bought it lol, but I works great connected via USB.

mwisniewski, 2024-01-20:
Hi, any updates on that topic?

tannewt, 2024-01-22:
From the [weekly meeting notes](https://github.com/adafruit/adafruit-circuitpython-weekly-meeting/blob/main/2024/2024-01-22.md#1914-status-updates):

> @eightycc (text only)
> 
> * I’ve been a bit under the weather for the last couple of weeks.
> * Continuing work on _bleio for RP2. Sorry this is taking so long, every bug I get past seems to have a dozen more hiding behind it. No idea how much longer this will take, but I am in it for the long haul.

This isn't easy work folks. Please sit tight.

foxt, 2024-01-23:
> then, I'd like to hear how community members plan to use _bleio and if possible give me code samples to work with. It's fine to use this issue to communicate with me.

I'd love to see [adafruit_ble_apple_notification_center](https://docs.circuitpython.org/projects/ble_apple_notification_center/en/latest/api.html) working on the Pico

alhafoudh, 2024-04-29:
Any news on this?

bablokb, 2024-05-03:
@eightycc Do you have a repo somewhere with your code? Your clone of CircuitPython on Github does not seem to contain your work on `_bleio`.  Maybe you could add a branch to your repo?! Making your efforts public will enable others to help.

alhafoudh, 2024-05-03:
I found a solution. Use Micropython and load in https://github.com/adafruit/Adafruit_Blinka to get the Circuitpython nice stuff. For BLE stuff use Micropython BLE code.

This helped me: https://www.youtube.com/watch?v=l254lxm78I4

hex-developer, 2024-05-04:
@alhafoudh that's a nice idea and would work in some cases, but I think most of the people interested in this issue would like to use some project which uses Adafruit's bluetooth libraries and wouldn't be easy to port. Or they might just want to write code that supports all CircuitPython boards, without requiring users to install firmware, libraries, etc.

KMK for example needs imports from `adafruit_ble` to work.

ItsTerm1n4l, 2024-07-01:
@sergi0g is exactly correct

ItsTerm1n4l, 2024-09-12:
@eightycc or @tannewt I understand that you are very busy but is there any update on this or any way to help?
Also how come there isn't a branch or fork with the features being worked on?
Thanks very much for your work.

tannewt, 2024-09-12:
> I understand that you are very busy but is there any update on this or any way to help?

No update about Pico W from me. BLE is supported on ESPs now in CircuitPython too. You could start working on it yourself too but its probably better to wait for @eightycc to reply.

> Also how come there isn't a branch or fork with the features being worked on?

It doesn't look like @eightycc has pushed their WIP code. It is up to them.

dmcomm, 2024-09-28:
Would Bluetooth support on Pico W be likely to reduce the available RAM again?

bablokb, 2024-09-28:
> With firmware CYW43 bluetooth firmware blobs, BTstack, and a skeleton implementation of _bleio in place flash grows from 85.7% to 88.14% and RAM from 40.14% to 44.35%. This includes both BLE and Wifi support.

From a post above in this now rather long thread.

@eightycc: is it possible that you publish the current state of your work somewhere so others might step in and continue on this project?

dmcomm, 2024-09-29:
> RAM from 40.14% to 44.35%

That's 11K, which could put it back into breaking projects that were broken by the 9.0 update and fixed again by disabling DVI and USB host recently.

ItsTerm1n4l, 2024-10-03:
@tannewt is @eightycc active on discord?

ItsTerm1n4l, 2024-10-03:
@jepler I know, I wanted to know if he was active because if he is, then I might have a look on the discord server to see if there are any updates there and if I could help in any way

tannewt, 2024-10-03:
I think it's safe to say that eightycc isn't available and that pinging them isn't helpful. If they were online, then they would have replied here by now. This happens from time to time as *volunteers* get busy.

I'm happy to guide anyone who wants to take on this work (though I'll soon be out as well for paternity leave.) Otherwise, there are plenty of other nRF52840 and ESP* boards that can already be used for BLE from CircuitPython.

AirplanegoBrr, 2024-10-24:
If Bluetooth isn't supported on the Pico, it should be removed from "Available on these boards" and a note should link back to this issue. Ridiculous.

[image]

tannewt, 2024-10-24:
> If Bluetooth isn't supported on the Pico, it should be removed from "Available on these boards" and a note should link back to this issue. Ridiculous.

It is available if you attach an AirLift ESP32.

dhalbert, 2024-10-24:
@AirplanegoBrr Showing anger doesn't make things happen. Your point about the documentation being ambiguous is taken, and I've opened a PR for that: #9758.

AirplanegoBrr, 2024-10-24:
> @AirplanegoBrr Showing anger doesn't make things happen. Your point about the documentation being ambiguous is taken, and I've opened a PR for that: #9758.

I just thought showing it on the compatibility list is a bit misleading- if you need extra hardware for something to work it should be mentioned but I guess that could mess up a lot of other boards and their documentation.

I apologize for getting upset in my last comment.

dhalbert, 2024-10-24:
> In the end it seems if you don't have nRF52840 then BLE just isn't going to work.

There is more than nRF52840. Status of current BLE implementations:
- native, on nRF52840 and nRF52833 boards
- native, on Espressif 8MB boards, except ESP32-S2, which doesn't have hardware support for BLE. Right now the extra code for BLE doesn't fit on 4MB boards because of an OTA update partition that is almost never used. In CircuitPython 10, we'll change the partition configuration and enable BLE on 4MB Espressif boards. But that requires a reformat (losing contents of CIRCUITPY) and also new TinyUF2 bootloaders. New bootloaders is a logistical thing we'll have to figure out how to deploy, label, etc.
- native, on SiLabs boards
- supported via AirLift ESP32 co-processors. Some boards have on-board AirLifts, like PyPortal, MatrixPortal M4, Metro M4 AirLift Lite. But also any boards with enough firmware space to include the HCI version of `_bleio` do so, so you can use an AirLift breakout board.

We concentrated on BLE for Espressif after nRF because it let us support BLE on many more boards and chips. Pico W was a single board, and the radio module used was not available separately. The module situation appears to be changing now.

bablokb, 2024-10-25:
I think part of the problem is that the Pico-W is *not* just another "single board". It is a board that has created a whole ecosystem. It is used in many educational projects. It is one of the few boards that work on two batteries with voltages down to 1.8V. Some manufacturers only produce RPxxxx based boards (Pimoroni is a prominent example). And like here in Germany, it is sometimes hard to buy quality boards from Adafruit, so if you want ESP32 boards, you have to order in China and the quality there gives you the whole range from perfect to unusable.

I think supporting the Pico-W is also a thing about reputation. I do read once in a while that CircuitPython is so much inferior to MicroPython because it does not manage to support a widely used chip that is already a few years old.

The message is (and you can read it multiple times in this thread): the users of CircuitPython really want BLE support for the Pico-W. I  know it is not your priority, but priorities can be changed.

ItsTerm1n4l, 2024-10-25:
just to add to this.
I think it is very useful to have a board that does what the pico does in that it is available in normal and wireless versions and it saves cost.
This does mean that if someone buys the pico W then they want the wireless capability.
Also it is useful to have a good all-round board with ON BOARD wifi instead of having to use a extra wifi board that adds to cost and complexity.
And the pico is quite popular so there would be many users that would use BLE on it if it was supported by circuitpython.

All that being said, I fully understand and appreciate that you are volunteers giving your time to help us and i thank you for that, but I would love to see this working on the pico or at the very least a branch or fork that we can all work on and collab to get this going, I'm not that great at coding but I want this feature bad enough that I would try to learn more and give it a shot :)

bfschris, 2024-10-28:
I'm relatively inexperienced and just started looking at the Pico W for a project which requires BLE support.  

The above thread left me a little confused, hoping someone will help me:

some say:  Pico W does not support BLE.  However, it seems MicroPython supports BLE on the Pico W; it's just CircuitPython which doesn't support that combination.  Is this true? and if so, what are the repercussions of using MicroPython instead of CP?

Or, is it possible to use a combination of the two, with the MP libraries used for BLE and everything else is CP?  I think I saw reference to that above, but wasn't sure it was verified or if someone was experimenting.

thanks very much -

ItsTerm1n4l, 2024-10-28:
@bfschris Circuitpython does not support BLE on the pico w, micropython does support BLE on the pico but there are some things that micropython cannot do and circuitpython is needed for, I also think circuitpython is more popular.
I dont think a combination is possible but someone else would have to verify that.
I hope this clears things up for you.

bfschris, 2024-10-28:
@ItsTerm1n4l  - thanks much, that does help.

Does anyone happen to know where I could find a comparison chart of the two?  or some kind of analysis of the pros/cons for the Pico W platform?  My goal being to see whether I can accept the reduced feature-set of micropython or whether the Pico W experiment is a dead-end for me.

thanks again -

[7 more replies]