adafruit/circuitpython #7703: OnDiskGif Memory Usage on SAMD51

gamblor21, 2023-03-10:
### CircuitPython version

```python
Adafruit CircuitPython 8.1.0-alpha.2-24-g26bab620f-dirty on 2023-03-02; Adafruit ItsyBitsy M4 Express with samd51g19
```

### Code/REPL

```python
import gifio
import gc

# Code ran at REPL
gc.collect()
gc.mem_free()
g = gifio.OnDiskGif("party_parrot_13.gif") # seems to need to be a larger size, exact size unknown
g.next_frame()
gc.mem_free()
del(g)
gc.mem_free()
gc.collect()
gc.mem_free()
```

### Behavior

Most times the output works as expected:
```
151744
0.04
95216
95216
151744
151744
```

With memory going back as expected. But sometimes you get a result like:
```
151744
0.04
95216
95216
95264
95264
```

Where the memory is not freed as expected (during runs the exact amount that was freed varied).

Running `gc.mem_free()` even after sitting at the REPL will show the memory not freed but after *some* time running `gc.collect()` and then `gc.mem_free()` will show the memory freed. Immediately running those commands pasted to the REPL is not enough time. Sometimes to trigger it just allocate a GIF twice in a row without collecting.

### Description

I could not get this to replicate on RP2040 or ESP32-S2, it seems to be SAMD51 only.

tannewt, 2023-03-10:
Things like this can happen if the pointer is stored on the stack and not overwritten.

tannewt, 2023-03-10:
Since you are editing the core and on M4 I'd add a global variable to store the pointer that the object to the object you think should be deleted. Once you have that, add an `asm('bkpt');` to `gc_mark` when the pointer passed in matches it. That'll stop gdb whenever the pointer is marked for keeping. The backtrace will show you what code found the pointer and where.

gamblor21, 2023-03-12:
I am still trying to figure out the *why* but sending the bitmap via the display bus seems to cause the issue. But if you assign the `OnDiskGif.bitmap` to a variable and pass that variable to the display_bus and then set that variable to None at the end the memory gets collected during the next run. Example below.

I am still going to try to look into why this happens (already learned a lot about garbage collection this weekend) but this is a pretty easy work around in the meantime. The example includes a call to `deinit()` which I added in PR  #7712 but even when commented out the collection error seems to go away.

```python
while True:
    for i in range(len(files)):
        if files[i][1] is True:  # Directory
            continue
        try:
            gc.collect()
            odg = gifio.OnDiskGif(files[i][0])
            gifbitmap = odg.bitmap
        except OSError:
            raise Exception("Can't open " + files[i][0])

        start = time.monotonic()
        next_delay = odg.next_frame()  # Load the first frame
        end = time.monotonic()
        call_delay = end - start

        end_time = time.monotonic() + 5

        while time.monotonic() < end_time:
            sleeptime = max(0, next_delay - call_delay)
            time.sleep(sleeptime)
            next_delay = odg.next_frame()
            display_bus.send(42, struct.pack(">hh", 0, odg.bitmap.width - 1))
            display_bus.send(43, struct.pack(">hh", 80, 80+odg.bitmap.height - 1))
            display_bus.send(44, gifbitmap)

        odg.deinit()
        gifbitmap = None
        del(odg)
```

dhalbert, 2023-03-12:
Maybe I am telling you something you know already: the garbage collector looks in several places, transitively, for references to storage not to reclaim:
1. "root pointers" (`MICROPY_PORT_ROOT_POINTERS`), which include the the globals dictionary, and many other things.
2. C variables handled by the various `gc_collect_*` routines. This is an alternative to making things root pointers.
3. The stack.

So perhaps the `bitmap` address is on the stack, or is left behind in something, like the `display_bus`.

gamblor21, 2023-03-13:
Thanks Dan, I had a vague idea of some of those points but not all together. Been trying to watch some of the past videos on memory management and looking into the code.

This is very new territory for me so I'm not 100% sure what I'm looking for besides the address of the bitmap somewhere on the stack or referenced by a root pointer. The breakpoint tip Scott gave me did get me to a breakpoint but it has been a lot of years since I did much with GDB.

gamblor21, 2023-03-13:
I confirmed that a pointer to the OnDiskGif.bitmap object was still on the stack, hence the GC would not clean the object up.

The issue stems from the fact you could get a reference to bitmap by calling it as a property vs a method call (`b = odg.bitmap` vs `b = odg.bitmap()`). When you load an attribute the function `mp_load_attr` is called vs `mp_load_method`. Load attribute allocates an array of mp_obj_t two entries long on the stack. Load method is similar but adds this to the top of the stack. The top almost inevitably gets overwritten.

Which is the other issue with objects being returned as properties. A non-object being returned will (almost) never match to something on the heap. An object has a good chance to match to something you are or recently used.

The fix I believe to make things work smoothest is to change the bitmap call to a method instead of a property. I already tested this on the SAMD51 and the issue went away.

I would also caution against any objects being returned as properties in the future.

dhalbert, 2023-03-14:
Is the allocated attribute on the used stack, or has that function returned? Sounds like the gc is not using the correct top of the stack, if the latter is true. Tagging @tannewt.

> The fix I believe to make things work smoothest is to change the bitmap call to a method instead of a property. I already tested this on the SAMD51 and the issue went away.

We should fix the real problem: this is a workaround rather than a fix.

gamblor21, 2023-03-14:
It is a reference on the stack that sometimes just is not overwritten and the stack is not zeroed out so when the gc runs it sees the pointer. For more detail:

in the interpreter when an attribute is loaded the main call is `obj = mp_load_attr(top, qst)` (vm.c:404). Withing `mp_load_attr` the first two lines are:
```c
mp_obj_t dest[2];
mp_load_method(base, attr, dest);
```

Putting `dest` onto the stack. In this case base will be an OnDiskGif object and attribute will be the bitmap attribute, telling the `mp_load_method` call we want the attribute named bitmap. When the call returns dest[0] holds a pointer to the heap where the `Bitmap` object exists. So far so good.

When we return to the VM the return value is stored in `obj` which was allocated by the VM on the stack and the stack is moved to point at that value. *but* dest[0] was still on the stack and set as a pointer to our bitmap. This is the memory field that unless it gets zeroed out later will trip up the gc. For non-object attributes the gc will see the value and ignore them as they do not point to heap memory (unless you're unlucky). It is only when we load an object as a property does this occur. This is also why if you load bitmap.height after loading the bitmap object itself the issue goes away. The property for height is on the stack still but isn't an object to be GCed.

Contrast this to loading a method with in vm.c:410
```c
mp_load_method(*sp, qst, sp);
sp += 1;
```
`mp_load_method`, the same one called from within `mp_load_attr`, does not allocate an array like `dest`. It instead stores the two values overwriting the current top of stack (which is no longer needed) and the next memory address. This value will later be popped off the stack to make the function call. 

All this only happened with `-Os` set. I'm guessing the compiler is making some guesses to shrink the size down about how the stack is used. It isn't a bug in how it is work. But a quirk of how the micropython gc works. I just watched Jim's youtube video talking about it (https://www.youtube.com/watch?v=H_xq8IYjh2w for anyone who wants to, its really informative) and he admits that items could be left on the stack. It isn't ideal but for speed and memory needed.

All this really matters because the large GIFs take up almost all the RAM on the M4 chip so an immediate GC collect is required to not run out of memory. And if something doesn't zero out that value on the stack it never will be freed.

I probably already wrote too much, but if anyone wants I can find time to talk about it on discord in voice if that is easier or can screen share a debugging session.

tannewt, 2023-03-14:
You could add `.deinit()` to Bitmap so that we can control when its memory is freed. Then this dangling reference will just be to the deinited Bitmap. Probably worth doing for something that can hold a bunch of memory.

gamblor21, 2023-03-14:
> You could add `.deinit()` to Bitmap so that we can control when its memory is freed. Then this dangling reference will just be to the deinited Bitmap. Probably worth doing for something that can hold a bunch of memory.

Sounds like a good idea. I will add the deinit to Bitmap in. Probably easier then tweaking the VM :)

gamblor21, 2023-03-24:
Fixed in #7712

closed 2023-03-24 by gamblor21