adafruit/circuitpython #7709: Changes in `bindings` directory are ignored

microdev1, 2023-03-10:
See: https://github.com/adafruit/circuitpython/actions/runs/4377633512/jobs/7661355349#step:13:1

microdev1, 2023-03-15:
Log of the `set-matrix` step:
```python
Run python3 -u ci_set_matrix.py
  python3 -u ci_set_matrix.py
  shell: /usr/bin/bash -e {0}
  env:
    pythonLocation: /opt/hostedtoolcache/Python/3.11.2/x64
    PKG_CONFIG_PATH: /opt/hostedtoolcache/Python/3.11.2/x64/lib/pkgconfig
    Python_ROOT_DIR: /opt/hostedtoolcache/Python/3.11.2/x64
    Python2_ROOT_DIR: /opt/hostedtoolcache/Python/3.11.2/x64
    Python3_ROOT_DIR: /opt/hostedtoolcache/Python/3.11.2/x64
    LD_LIBRARY_PATH: /opt/hostedtoolcache/Python/3.11.2/x64/lib
    CP_VERSION: 8.1.0-beta.0-16-g63638a56f
    HEAD_SHA: 58f28b98be93a25ebe1d75fbfff2b4a1ebff7009
    BASE_SHA: 874ba4ec68dae62914d04edb543dd90bf98be8ba
    CHANGED_FILES: ["ports/espressif/bindings/espnow/ESPNow.c"]
    LAST_FAILED_JOBS: 
Using files list in CHANGED_FILES
LAST_FAILED_JOBS is in environment, but value is empty
Adding docs/boards to build based on changed files
Building docs: True
Building windows: False
Building boards: False
```

dhalbert, 2023-03-15:
This seems odd: I would think any change below `ports/espressif` would work. Do you have an idea of the problem?

dhalbert, 2023-03-15:
It's not just `bindings`, I think maybe it's `CIRCUITPY_*` features that are not across all parts.
```sh
$ tools/ci_set_matrix.py ports/espressif/bindings/espnow/ESPNow.c
...
Building boards: False
...
```
```sh
$ tools/ci_set_matrix.py ports/espressif/common-hal/espnow/ESPNow.c
...
Building boards: False
...
```
```sh
$ tools/ci_set_matrix.py ports/espressif/common-hal/busio/UART.c
...
Building boards: True
  adafruit_esp32s2_camera
  adafruit_feather_esp32_v2
  adafruit_feather_esp32s2
...
```

microdev1, 2023-03-15:
The module is searched in `SRC_PATTERNS` but the modules under `bindings` aren't in it.
https://github.com/adafruit/circuitpython/blob/37b9f778e1017b1cc8704f50ed13a04414e20b4b/tools/ci_set_matrix.py#L215-L220

PR #7762 (2023-03-21): CI: Update set matrix script
In this PR:

- Add `espressif` specific modules to `SRC_PATTERNS`. Closes #7709.
- Don't run all jobs if changed files is empty.
closed 2023-03-21 by dhalbert