adafruit/circuitpython #7716: ESP32-C3 Serial instability

bill88t, 2023-03-11:
This has been a long standing issue that just nobody has bothered with.
It applies to all CP versions.

Under several circuimstances, CircuitPython on ESP32-C3 decides to lock up stdin or go out of sync.

Some ways to reproduce it:
- Press any key before before circuitpython has finished init.
- Press too many keys at once.
- Work with serial for more than 3 minutes.
- Use ljinux. It very heavily relies on ANSI escapes for graphics.

Ctrl+C usually makes it through regardless, but doesn't clear stdin (this could be used as a workaround).

Too many Ctrl+C at once send garbage:
```
[board@ljinux| ~]> [[[[[5;20R[[^C
> ;ins5;ins5ins5ins5;20Rins5ins5ins5ins5ins5ins5ins5ins5ins5ins5ins5ins5ins5;20R
```

In some rare cases stdin goes out of sync instead of locking up.
What this means is that I type `abcdef`, `abc` is echo'ed back and then if I press `ghi`, `def` is echo'ed.

RetiredWizard, 2024-03-28:
@bill88t Was this resolved by #7973?

bill88t, 2024-03-28:
There is still a oddly specific Cx desync issue which I have workarounds in place for.
With that one workaround it's perfectly stable.
Without it, my terminal line is not printing no matter how hard I `.flush()`.

gitcnd, 2024-07-02:
>  no matter how hard I .flush().

*what* exactly, are you flushing?

`sys.stdout.flush() # AttributeError: 'FileIO' object has no attribute 'flush'`

Same problem has been driving me *crazy* on both my ESP32-C3 (Qt Py) and my ESP32-S2

bill88t, 2024-07-02:
> what exactly, are you flushing?

Oh I am using the [jcurses](https://github.com/beryllium-org/jcurses) library and [virtual consoles](https://github.com/beryllium-org/virtUART) on Cx and regular esp32 for interoperability between boards.
Kinda forgot about that. Yea flushing has no effect there.
Oh and btw the workaround to the bug is sending ansi escapes. `b"\x1b[s\x1b[1B\x1b[1C\x1b[u"` works really well.