adafruit/circuitpython #7720: unable to use shell with 8.0.3 but OK with 8.1.0-betat

jerryneedell, 2023-03-13:
I'm not sure this is really an "issue" since it appears to have been resolved in 8.1.0-beta.0 but I wanted to bring it to your attention.
A forum post reported an issue using rshell with 8.0.3. on a feather esp32 v2  In general, you do have to download and install shell from GitHub to get a necessary fix that has been merged but not released. I shoed the poster that shell did work OK with 8.1.0-beta.0 however they still reported an error with 8.0.3. I then reverted to 8.03 and was able to reproduce the error.
After reinstalling 8.1.0-beta.0 it now works again.

Here is a link to the forum post https://forums.adafruit.com/viewtopic.php?p=964655#p964655

It is not obvious to me what "fixed" this in 8.1.0-beta.0 but I though it worth reporting.

jerryneedell, 2023-03-13:
Thanks -- No need to keep this open.

closed 2023-03-13 by jerryneedell

tannewt, 2023-03-14:
@Neradoc should we make it not translated?