adafruit/circuitpython #7721: Build for esp32 and pico_w throws error in 'gen_web_workflow_static.py' import 'minify_html'

kreier, 2023-03-13:
### CircuitPython version

```python
Adafruit CircuitPython 8.1.0-beta.0-4-g8a1006999 on 2023-03-13; LilyGO T-Display with rp2040
```

### Code/REPL

```python
# Don't get that far - compiling for rp2040 works but not if the web workflow is involved
# see below
```

### Behavior

mk@zbook:~/circuitpython_esp32/ports/raspberrypi$ make BOARD=raspberry_pi_pico_w
Use make V=1, make V=2 or set BUILD_VERBOSE similarly in your environment to increase build verbosity.
mkdir -p build-raspberry_pi_pico_w/genhdr
Traceback (most recent call last):
  File "../../tools/gen_web_workflow_static.py", line 7, in <module>
    import minify_html
ModuleNotFoundError: No module named 'minify_html'
make: *** [../../supervisor/supervisor.mk:175: build-raspberry_pi_pico_w/autogen_web_workflow_static.c] Error 1

mk@zbook:~/circuitpython_esp32/ports/espressif$ make BOARD=adafruit_metro_esp32s2
Use make V=1, make V=2 or set BUILD_VERBOSE similarly in your environment to increase build verbosity.
mkdir -p build-adafruit_metro_esp32s2/genhdr
Traceback (most recent call last):
  File "../../tools/gen_web_workflow_static.py", line 7, in <module>
    import minify_html
ModuleNotFoundError: No module named 'minify_html'
make: *** [../../supervisor/supervisor.mk:175: build-adafruit_metro_esp32s2/autogen_web_workflow_static.c] Error 1

### Description

Compiling works for stm32 and rp2040 but not when the web workflow is involved. I created a new fresh folder for esp32 but as the above pictures show the error message is the same. Planned target was an esp32. But both esp32s2 and pico_w (rp2040 with wifi) fail.

kreier, 2023-03-13:
The import is in line 7 of https://github.com/adafruit/circuitpython/blob/main/tools/gen_web_workflow_static.py 

`pip install minify_html` and `pip install jsmin` solved the problem for the Raspberry Pico W.

esp32 still fails with `CMake Error: CMake was unable to find a build program corresponding to "Ninja".  CMAKE_MAKE_PROGRAM is not set.  You probably need to select a different build tool.`

kreier, 2023-03-13:
Same error for `mk@zbook:~/circuitpython_rp2040/ports/espressif$ make BOARD=adafruit_metro_esp32s2`
"Ninja" can't be found.

closed 2023-03-13 by kreier

kreier, 2023-03-13:
CMake was unable to find a build program corresponding to "Ninja". 

sudo apt install ninja-build

New error:

COMPONENTS = esptool_py;soc;driver;log;main;esp-tls;mbedtls;mdns;esp_event;esp_adc_cal;esp_netif;esp_wifi;lwip;ulp;wpa_supplicant;freertos;bt;usb;esp32-camera
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
-- The ASM compiler identification is unknown
-- Found assembler: xtensa-esp32s2-elf-gcc
CMake Error at esp-idf/tools/cmake/project.cmake:296 (__project):
  The CMAKE_C_COMPILER:

    xtensa-esp32s2-elf-gcc

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.
Call Stack (most recent call first):
  CMakeLists.txt:19 (project)

CMake Error at esp-idf/tools/cmake/project.cmake:296 (__project):
  The CMAKE_CXX_COMPILER:

    xtensa-esp32s2-elf-g++

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.
Call Stack (most recent call first):
  CMakeLists.txt:19 (project)

CMake Error at esp-idf/tools/cmake/project.cmake:296 (__project):
  The CMAKE_ASM_COMPILER:

    xtensa-esp32s2-elf-gcc

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "ASM" or the CMake cache entry CMAKE_ASM_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.
Call Stack (most recent call first):
  CMakeLists.txt:19 (project)

-- Warning: Did not find file Compiler/-ASM
-- Configuring incomplete, errors occurred!

microdev1, 2023-03-13:
Hi, @kreier replying to the error in your last comment, it looks like you didn't run `source esp-idf/export.sh` in the `ports/espressif` directory. Please check the [esp-idf getting started guide](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/index.html) for more details.

dhalbert, 2023-03-13:
@kreier Did you see https://learn.adafruit.com/building-circuitpython/espressif-build ?