adafruit/circuitpython #7722: watchdog: make behavior across ports consistent

jepler, 2023-03-13:
[based in part on discussion In The Weeds on the weekly meeting March 13, 2023]

 * In shared-bindings permit assigning `watchdog.mode = None` if the underlying hardware supports it
 * In a later version, remove the `Watchdog.deinit()` method. (today, calling this is the workaround for setting watchdog.mode to None)
 * do something to make the behavior of watchdog at the end of code.py consistent; probably, only inactivate the watchdog in the case of KeyboardInterrupt or ReloadException rather than at any interpreter exit (`watchdog_reset()`)

vladak, 2024-03-08:
What's the story behind the removal of `deinit()` ? In my case, I call it before the device (Feather V2) is put to deep sleep, which happens periodically. Maybe calling it is not necessary in such case ?

microdev1, 2024-03-08:
> What's the story behind the removal of `deinit()` ?

The `deinit` functionality is still present, the API has changed. Instead of calling `deinit()`, set the watchdog mode to `None`.

PR #10144 (2025-03-19): Remove watchdog.deinit() for 10.0.0.
Removes deprecated `watchdog.deinit()` for CP 10.0.0.

Resolves #7722.
closed 2025-03-19 by tannewt