adafruit/circuitpython #7726: SPI Malfunction

default-settings, 2023-03-14:
### CircuitPython version

```python
Version 8.0.0 on PCA10100
```

### Code/REPL

```python
import busio, board, digitalio, time
from adafruit_bus_device.spi_device import SPIDevice

buf = bytes([0x90,0x00,0x00,0x00])

cs = digitalio.DigitalInOut(board.P0_17)
comm_port = busio.SPI(board.P0_14, MOSI=board.P0_20, MISO=board.P0_21)
device = SPIDevice(comm_port, cs)

with device as bus_device:
    bus_device.write(buf)
    result = bytearray(2)
    bus_device.readinto(result)

print("".join("{:02x}".format(x) for x in result))
```

### Behavior

[image]
[image]

Device becomes undetectable. CircuitPython Drive disappears from file explorer.

### Description

Also tried using general busio.write() and not SPIDevice. Same error. I have to reflash the board each time.

kreasteve, 2023-03-14:
Why do you call it comm_port?

Don't forget `cs.direction = digitalio.Direction.OUTPUT` to set the Cs pin to output.

dhalbert, 2023-03-14:
@default-settings:

1. Did you flash the UF2 bootloader and the SoftDevice to the board?
2. If you remove the print statement, does it still fail?

I tried this code (just with different pins) on a Feather nRF52840 with 8.0.3, and it works.

default-settings, 2023-03-14:
@kreasteve 
I tried that when I wasn't using SPIDevice and just busio and manually setting it to `False`. I named it comm_port because this is an almost copy/paste from the example test from SPIDevice.
@dhalbert 
I've flashed the bootloader and the soft device, yes. Removing the print did nothing, it fails during the write. I was able to lock and unlock the bus, but the write makes it fail.

dhalbert, 2023-03-17:
Reproduced on a PCA10100 eval board (nrf52833), using SoftDevice 7.2.0, installed by the latest Adafruit_nRF52_Bootloader.

The first SPI assigned is probably SPIM3. By not using the first SPI assigned, I worked around the problem. This code does not hang:
```py
import busio, board, digitalio, time
from adafruit_bus_device.spi_device import SPIDevice

buf = bytes([0x90,0x00,0x00,0x00])

cs = digitalio.DigitalInOut(board.D3)
# comm_portx will get SPIM3
comm_portx = busio.SPI(board.D0, MOSI=board.D1, MISO=board.D2)
# comm_port will get another SPIM
comm_port = busio.SPI(board.D4, MOSI=board.D5, MISO=board.D6)
device = SPIDevice(comm_port, cs)

with device as bus_device:
    bus_device.write(buf)
    result = bytearray(2)
    bus_device.readinto(result)

print("".join("{:02x}".format(x) for x in result))
```

dhalbert, 2023-03-17:
it is possible to avoid assigning SPIM3 at all: change `nrfx_config.h` to not use it:
```c
-- a/ports/nrf/nrfx_config.h
+++ b/ports/nrf/nrfx_config.h
@@ -44,7 +44,7 @@
 #ifndef NRFX_SPIM3_ENABLED
 #if defined(NRF52840_XXAA) || defined(NRF52833_XXAA)
     #define NRFX_SPIM_EXTENDED_ENABLED 1
-    #define NRFX_SPIM3_ENABLED 1
+    #define NRFX_SPIM3_ENABLED 0
 #elif CIRCUITPY_NRF_NUM_I2C == 2
     #define NRFX_SPIM3_ENABLED 0
 #endif
```
There should be a cleaner way to do this in `mpconfigboard.h`, but there is not right now.

default-settings, 2023-03-20:
Perfect! That fixed the issue

closed 2023-03-20 by default-settings

dhalbert, 2023-04-14:
Could not assign SPIM3 unless the speed is >8MHz, and then reassign the device to that in that case.

mrdalgaard, 2023-04-18:
Related?
https://github.com/adafruit/circuitpython/issues/5233

dhalbert, 2023-04-18:
> Related? #5233

In this particular case, USB is connected, so I don't think it's directly #5233. However, SPIM3 is flaky in various ways, so in the long run, I think we should avoid assigning it unless the user really wants/needs it.