adafruit/circuitpython #7727: The module contents are repeated in the tree

microdev1, 2023-03-15:
Is this the intended behavior?
   
[image]

tannewt, 2023-03-16:
I don't think that's intended.

microdev1, 2023-03-29:
Did something change? The non-boxed versions have disappeared.

Neradoc, 2023-03-29:
Comparing the logs of the last run and one a day ago, I see sphinx autoapi 2.1.0 has been released.
And this PR might be relevant:
https://github.com/readthedocs/sphinx-autoapi/pull/374

> Sphinx 5.2.0 adds domain objects to the toctree by default. This is causing duplicate entries to appear in the table of contents when autoapi is in use, since autoapi also adds domain objects to the toctree by default (e.g. #369). It seems sensible to allow sphinx to handle the addition of domain objects to the toctree instead of post-processing it ourselves in autoapi.
> 
> If merged, this PR will cause autoapi to defer to sphinx to handle adding domain objects to the toctree.

microdev1, 2023-03-29:
> This last PR merged is relevant: https://github.com/readthedocs/sphinx-autoapi/pull/374

Okay, so this can be closed now.

PR #7278 (2022-11-29): Sdcardio failures
Before this, CircuitPython was unusable on the SAM E54 XPLAINED devkit. After this, I'm able to initialize the following cards successfully:
 * Lexar Platinum II 32GB full-sized SD HC card
 * unbranded 2GB microSD card
 * Kingston 2GB microSD HC card

This is using the built in full sized SD card slot on the devkit.

Before the second change, after any attempt to construct an sdioio.SDCard instance (failed or successful) the pins would remain "in use" and a new instance could not be constructed. Now, an SDCard instance can be deinitialized with `deinit()` or by reset of the i [... cut]
closed 2023-03-29 by microdev1