adafruit/circuitpython #7732: Storage.VfsFat.mkfs() don't work

kreasteve, 2023-03-16:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.4 on 2023-03-15; Adafruit Feather RP2040 with rp2040
```

### Code/REPL

```python
import sdcardio
import storage
import board

spi = board.SPI()
cs = board.D10

sdcard = sdcardio.SDCard(spi, cs)
vfs = storage.VfsFat(sdcard)
vfs.mkfs()
```

### Behavior

`vfs.mkfs()` throws TypeError: function takes 1 positional arguments but 0 were given.

### Description

I want to be able to format my sd-card, but maybe I misunderstand the `mkfs()` method.

https://docs.circuitpython.org/en/latest/shared-bindings/storage/index.html#storage.VfsFat.mkfs

kreasteve, 2023-03-16:
Hi Dan!

I've tried your code on the rp2040.

the line `storage.VfsFat.mkfs(sdcard)`
throws: OSError: [Errno 22] Invalid argument

Is that vfs_fat.c in extmod?

dhalbert, 2023-03-16:
> Is that vfs_fat.c in extmod?

Yes, it is.

Here is my actual transcript on the PyPortal (removing some typing errors). I will try it on an RP2040 later. Which board and SD breakout (if any) are you using?
```py
>>> import sdcardio,storage,board
>>> spi = board.SPI()
>>> sdcard = sdcardio.SDCard(spi,board.SD_CS)
>>> storage.VfsFat.mkfs(sdcard)
>>> import os
>>> vfs = storage.VfsFat(sdcard)
>>> storage.mount(vfs, "/foo")
>>> os.listdir()
['foo', '.fseventsd', '.metadata_never_index', '.Trashes', 'sdtest.py', 'lib', 'boot_out.txt', 'foo.txt', 'code.py', 'bar.txt', 'System Volume Information', 'secrets.py', 'wifi.py', 'codex.py', 'this-is.gif', 'nyan.gif']
>>> os.listdir("/foo")
[]
```

kreasteve, 2023-03-16:
It is a feather rp2040
an adalogger featherwing and an oled wing

dhalbert, 2023-03-16:
Could you check your code again?  Here is what I tried from the REPL, with a Feather RP2040 and an AdaLogger FeatherWing:
```py
>>> import sdcardio,storage,board,os
>>> spi = board.SPI()
>>> sdcard = sdcardio.SDCard(spi,board.D10)
>>> storage.VfsFat.mkfs(sdcard)
>>> vfs = storage.VfsFat(sdcard)
>>> storage.mount(vfs, "/foo")
>>> os.listdir("/foo")
[]
```
I tried this with 8.0.4 and 8.1.0-beta.0, and it worked for both.

kreasteve, 2023-03-16:
No sorry, got the same error. With 8.1.0 too. Now without the oled featherwing.

```python
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== import sdcardio,storage,board,os
=== spi = board.SPI()
=== sdcard = sdcardio.SDCard(spi,board.D10)
=== storage.VfsFat.mkfs(sdcard)
=== vfs = storage.VfsFat(sdcard)
=== storage.mount(vfs, "/foo")
=== os.listdir("/foo")
Traceback (most recent call last):
  File "<stdin>", line 4, in <module>
OSError: [Errno 22] Invalid argument
```
I can read and write the sd card. But I can't use mkfs.

dhalbert, 2023-03-16:
What is the brand and capacity of the SD card? I am using a 512MB card.

kreasteve, 2023-03-16:
It is Samsung Evo+ 32GB, thats what I had laying around. But I can try 4GB (other brand) in a few days.

I also could try this tomorrow with a new set of boards. Maybe the devil is in the details.

dhalbert, 2023-03-16:
I think it is whether it can create the right kind of FAT. It is too big for regular FAT32.

If you format this card on a host computer, and then just use it, does it work OK?

kreasteve, 2023-03-17:
**My log:**
It is `FR_INVALID_PARAMETER`. I've replaced the 3 occurences of this number with 3 different numbers (1,2,3).

The error occurs here:
`if (!(opt & FM_FAT)) LEAVE_MKFS(FR_INVALID_PARAMETER);  /* no-FAT? */` (lib/oofatfs/ff.c)

`FF_MKFS_FAT32` is 0. Which causes this "bug"

**How I get it working:**
```c
 #ifdef MICROPY_FF_MKFS_FAT32
 #define FF_MKFS_FAT32   MICROPY_FF_MKFS_FAT32
 #else
-#define FF_MKFS_FAT32   0
+#define FF_MKFS_FAT32   1
 #endif
 /* This option switches off FAT32 support in f_mkfs() */
```
`MICROPY_FF_MKFS_FAT32` is not defined. This leads to FAT32 is not enabled.

With this little workaround I'm able to format my 32GB sd card in about 5 seconds.

I just want to know, where the error comes from. This may not be the solution, but maybe it helps you to fix this.

jepler, 2023-03-17:
I've entered a PR which should re-enable mkfs for >4GB filesystem on most builds, including ones for rp2040. (Some very resource constrained boards, mostly samd21, would still lack support for this) -- See #7739. Once there are build artifacts, your feedback from testing with them would be helpful.

PR #7739 (2023-03-17): re-enable mkfs for >4GB filesystems whenver FULL_BUILD
We (I) disabled this across the board to save firmware size for small SAMD boards. Re-enable it everywhere where we have "full build"s.

According to my notes, 12c95ee9e96e9c5fd3609a6f095cd80fc2a849f3 saved 508 bytes of flash.

Closes: #7732
closed 2023-03-17 by dhalbert