adafruit/circuitpython #7738: Re-overwrite the file. The history file still exists

xcg340122, 2023-03-17:
### CircuitPython version

```python
8.0.4
```

### Code/REPL

```python
import storage

# 禁用 USB MSC
storage.disable_usb_drive()
```

### Behavior

The drive is gone.

### Description

After I burned the firmware for Raspberry Pi Pico using uf2 for version circuitpython8.0.4, I created boot.py and disabled the drive with storage.disable_usb_drive(). Now I want to redisplay the drive, After I tried to re-install circuitpython by pressing the BOOTSEL button on the motherboard, I found that the drive still did not appear. Interestingly, I could see the list of files through Thonny's IDE, but I couldn't delete or modify them

microdev1, 2023-03-17:
The filesystem persists on a firmware re-flash. You can re-write `boot.py` or erase fs using `storage.erase_filesystem()` from repl.

closed 2023-03-17 by microdev1

xcg340122, 2023-03-17:
After I re-copy flash_nuke.uf2, the reset is successful, thank you