adafruit/circuitpython #7742: Beta Version 8.1.0 DHT11 not found

electromechpro, 2023-03-17:
### CircuitPython version

```python
Adafruit Circuitpython 8.1.8 on 2023-03-17; Adafruit Metro ESP32S2 with ESP32S2
```

### Code/REPL

```python
# SPDX-FileCopyrightText: 2017 Limor Fried for Adafruit Industries
#
# SPDX-License-Identifier: MIT

import time
import adafruit_sdcard
import board
import busio
import digitalio
import adafruit_dht
import microcontroller
import storage
import adafruit_pcf8523

# Use any pin that is not taken by SPI
SD_CS = board.IO9
# Initial the dht devices, with data pin connected to:
dhtDevice1 = adafruit_dht.DHT11(board.IO10)
dhtDevice2 = adafruit_dht.DHT11(board.IO11)
led = digitalio.DigitalInOut(board.LED)
led.direction = digitalio.Direction.OUTPUT
fan = digitalio.DigitalInOut(board.A2)
fan.direction = digitalio.Direction.OUTPUT

# Connect to the card and mount the filesystem.
spi = busio.SPI(board.SCK, board.MOSI, board.MISO)
cs = digitalio.DigitalInOut(SD_CS)
sdcard = adafruit_sdcard.SDCard(spi, cs)
vfs = storage.VfsFat(sdcard)
storage.mount(vfs, "/sd")
myI2C = busio.I2C(board.SCL, board.SDA)
rtc = adafruit_pcf8523.PCF8523(myI2C)
days = ("Sunday", "Monday", "Tuesday", "Wednesday", "Thursday", "Friday", "Saturday")

# Use the filesystem as normal! Our files are under /sd

print("Logging temperature to filesystem")
# append to the file!
while True:
    # Print the values to the serial port
    temperature_c1 = dhtDevice1.temperature
    temperature_f1 = temperature_c1 * (9 / 5) + 32
    humidity1 = dhtDevice1.humidity
    temperature_c2 = dhtDevice2.temperature
    temperature_f2 = temperature_c2 * (9 / 5) + 32
    humidity2 = dhtDevice2.humidity
    timestamp = rtc.datetime
    delta_T = "%.1f F\t" %(temperature_f1 - temperature_f2)
    DT_int = temperature_f1 - temperature_f2
    if DT_int > 2:
        fan.value = True
        print("Fan ON due to Delta T")
    else:
        fan.value = False
        print("Fan OFF now")
    print("%d:%d %d/%d/%d" %(timestamp.tm_hour, timestamp.tm_min, timestamp.tm_mon, timestamp.tm_mday, timestamp.tm_year))
    timestring = "%d:%d %d/%d/%d\t" %(timestamp.tm_hour, timestamp.tm_min, timestamp.tm_mon, timestamp.tm_mday, timestamp.tm_year)

    print(
        "Inside Temp: {:.1f} F / {:.1f} C    Humidity: {}% ".format(
            temperature_f1, temperature_c1, humidity1
        )
    )
    print(
        "Outside Temp: {:.1f} F / {:.1f} C    Humidity: {}% ".format(
            temperature_f2, temperature_c2, humidity2
        )
    )
    # open file for append
    with open("/sd/temp&humid.txt", "a") as f:
        led.value = True  # turn on LED to indicate we're writing to the file
        c = microcontroller.cpu.temperature
        f.write(timestring)
        f.write("Inside Temp: {:.1f} F / {:.1f} C    Humidity: {}% ".format(temperature_f1, temperature_c1, humidity1))
        f.write("Outside Temp: {:.1f} F / {:.1f} C    Humidity: {}% ".format(temperature_f2, temperature_c2, humidity2))
        f.write("   Delta T: ")
        f.write(delta_T)
        f.write("   CPU Temp: %0.1f\n" % c)
        led.value = False  # turn off LED to indicate we're done
    # file is saved
    time.sleep(600)
```

### Behavior

DHT11 not found

(in line 200 something of adafruit_dht)

I have a Metro ESP32-S2 running 2 DHT-11 sensors on IO10 and 11, with an Arduino brand ethernet shield v2 acting as a datalogger for inside/outside temps. I was having no luck trying to add a soft-AP for monitoring it (remote site, no wifi), following the git hub and stack exchange threads. I tried installing CP ver 8.1 Beta, and still no AP love.

When I went back to my DHT datalogger code, the CP (8.1) code would not pick up the 2 DHT sensors (I was not able to figure which one or both, and swapped in a known good one at either location, no difference).

I then used the Adafruit web serial flasher and reverted back to 8.0.3, and with no code changes, the DHTs gave valid temperature readings.

I apologize for my clumsy coding.  I am new to CP. 

### Description

Error started when flashing from CP 8.0.3 to 8.1.0, then went away when reverting to 8.0.3

electromechpro, 2023-03-17:
Additional information: Adafruit RTC PCF8523 on I2C port of Arduino Ethernet Shield v2

electromechpro, 2023-04-04:
As of 4/3/23, version 8.1.0 for the Metro ESP32-S2 still has this DHT issue, at least with the DHT11.  V8.0.3 still works with DHT11.

DavePutz, 2023-04-06:
This is a regression caused by https://github.com/adafruit/circuitpython/commit/1063ec064961ab40121f1aebf2746de09a6cedcf
moving port_background_task() to port_background_tick(); so pulsein_background() is no longer getting called. Looking into possible fix.

closed 2023-04-10 by microdev1