adafruit/circuitpython #7743: Include example of how alarm.time() is used in code.py

kattni, 2023-03-17:
The documentation for `alarm.time()` should include an example that shows what it looks like in use in `code.py`. I struggled to figure out how to format it, and ultimately asked for help. I believe this would eliminate potential confusion for others moving forward.

The following explanation and example should be fine to include in the `alarm.time()` documentation.

This code block sets a `delay`, creates a `TimeAlarm` to wake up the board after `delay` seconds have passed, and then deep sleeps the board until the time alarm is triggered in 30 seconds.

```py
delay = 30  # delay is in seconds
time_alarm = alarm.time.TimeAlarm(monotonic_time=time.monotonic() + delay)
alarm.exit_and_deep_sleep_until_alarms(time_alarm)
```

AkshayaVarshieni14, 2023-03-18:
im ready to do this

deshipu, 2023-07-29:
@AkshayaVarshieni14 are you still working on it, or is this issue free to be taken by someone else?

dhalbert, 2025-03-22:
Fixed by #10160.

PR #10160 (2025-03-21): Fix multiple documentation issues.
- Fixes #10037.
- Fixes #10147.
- Fixes #9690.
- Fixes #9733.
- Fixes #7943.
- Fixes #7743.
- Fixes #4493.

Also fixes a few minor typos I encountered elsewhere.
closed 2025-03-22 by dhalbert