adafruit/circuitpython #7747: PWMOut reports "All timers in use" after creating a variable_frequency PWMOut object

RetiredWizard, 2023-03-18:
### CircuitPython version

```python
Adafruit CircuitPython 8.1.0-beta.0-33-g9c1d83476 on 2023-03-17; Adafruit Grand Central M4 Express with samd51p20
```

### Code/REPL

```python
import board
import pwmio
a = pwmio.PWMOut(board.D8)
a.deinit()
a = pwmio.PWMOut(board.D8)
a.deinit()
a = pwmio.PWMOut(board.D7,variable_frequency=True)
a.deinit()
a = pwmio.PWMOut(board.D8)
```

### Behavior

```python
>>> import board
>>> import pwmio
>>> a = pwmio.PWMOut(board.D8)
>>> a.deinit()
>>> a = pwmio.PWMOut(board.D8)
>>> a.deinit()
>>> a = pwmio.PWMOut(board.D7,variable_frequency=True)
>>> a.deinit()
>>> a = pwmio.PWMOut(board.D8)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: All timers in use
```

### Description

Once a PWMOut object is created using the variable_frequency=True option some pins which could previously have PWMOut objects successfully created on them return an "All timers in use" error. A Control-D does not clear the issue, a board reset is required to allow the erring pins to once again be used to create a PWMOut object. Only some pins seems to be impacted, others continue to function as expected.

### Additional information

By using the script from the learning guide:
```python
# SPDX-FileCopyrightText: 2018 Kattni Rembor for Adafruit Industries
#
# SPDX-License-Identifier: MIT

"""CircuitPython Essentials PWM pin identifying script"""
import board
import pwmio

for pin_name in dir(board):
    pin = getattr(board, pin_name)
    try:
        p = pwmio.PWMOut(pin)
        p.deinit()
        print("PWM on:", pin_name)  # Prints the valid, PWM-capable pins!
    except ValueError:  # This is the error returned when the pin is invalid.
        print("No PWM on:", pin_name)  # Prints the invalid pins.
    except RuntimeError:  # Timer conflict error.
        print("Timers in use:", pin_name)  # Prints the timer conflict pins.
    except TypeError:  # Error returned when checking a non-pin object in dir(board).
        pass  # Passes over non-pin objects in dir(board).
```
before and after creating a variable_frequency PWMOut object all the pins affected can be identified.

tannewt, 2023-03-20:
Yes, `variable_frequency` can cause other pins to not find a timer. Timers are only shared when the base frequency is fixed.

A reload should clear any pending state. If it doesn't, then it's a bug.

dhalbert, 2025-02-01:
I think this will have been fixed by #8966.

RetiredWizard, 2025-02-01:
I believe this falls into sort of a known issue or expected behavior category, so I'm fine closing it. 

That being said, from absolute newest:
```
Adafruit CircuitPython 9.2.4-4-g5d39e467fb on 2025-01-31; Adafruit Grand Central M4 Express with samd51p20
>>> import board
>>> import pwmio
>>> a = pwmio.PWMOut(board.D8)
>>> a.deinit()
>>> a = pwmio.PWMOut(board.D8)
>>> a.deinit()
>>> a = pwmio.PWMOut(board.D7,variable_frequency=True)
>>> a.deinit()
>>> a = pwmio.PWMOut(board.D8)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: Internal resource(s) in use
```
The error message has changed but PWMOut is still not happy using D8 after variable_frequency is used. A control-D doesn't free up the resources, however a power cycle will.

dhalbert, 2025-02-01:
I was hoping the finaliser change fixed this, surprised it didn't. Maybe `deinit()` is not clearing stuff. Reopening. Thanks for testing.

istvanzk, 2025-02-14:
Hi

I'm having the same issue when using a RP2350 based board, and on different GPIOs than above, not used by anything else in my case. Strangely, a full hardware reset does not help, so it indicates that the L1 pwmio object creation does not allow the creation of R2. Note that I'm using on purpose GPIOs corresponding to the same PWM slice for L1/R1 (slice #1) and L2/R2 (slice #5), as I do need them to share the clock frequency, but not the duty cycle later.

>>> import board, pwmio
>>> PWML1_Pin = board.GP2
>>> PWML2_Pin = board.GP24
>>> PWMR1_Pin = board.GP3
>>> PWMR2_Pin = board.GP25
>>> L1 = pwmio.PWMOut(PWML1_Pin, duty_cycle=0, frequency=20, variable_frequency=True)
>>> L2 = pwmio.PWMOut(PWML2_Pin, duty_cycle=0, frequency=20, variable_frequency=True)
>>> R1 = pwmio.PWMOut(PWMR1_Pin, duty_cycle=0)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: Internal resource(s) in use