adafruit/circuitpython #7769: intermitant code freeze using touch capacitance

teejay0909, 2023-03-22:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.4 on 2023-03-15; Adafruit Feather M0 Express with samd21g18
```

### Code/REPL

```python
import gc
import board
import adafruit_vl53l4cd  # TOF
import time
import neopixel
import simpleio
import touchio
from digitalio import DigitalInOut, Direction, Pull

gc.collect()   # make some rooooom, WIP, copied from code on M0 Express board

pixel_pin = board.NEOPIXEL
num_pixels = 1
ORDER = neopixel.GRB
pixels = neopixel.NeoPixel(
    pixel_pin, num_pixels, brightness=1, auto_write=False, pixel_order=ORDER
)
pixels.fill((0, 0, 100))  # (R,G,B) show distinct blue during boot up
pixels.show()

touch = touchio.TouchIn(board.A2)  # used to sleep MCU.

micro_com_out = DigitalInOut(board.D9)
micro_com_out.direction = Direction.OUTPUT
micro_com_out.value = True

i2c = board.I2C()  # uses board.SCL and board.SDA # updated for M0 Express
vl53 = adafruit_vl53l4cd.VL53L4CD(i2c)
vl53.inter_measurement = 0
vl53.timing_budget = 200
vl53.start_ranging()

max_acum = 10  # prevent hysteresis
acum_too_long = max_acum  # initialize to "full tube condition".
thd_IN = 1.3  # empty tube measure 1.3:1.4 inch.

sleep_acum = 0
sleep_max = 20
sleep_direction = 1
sleeping_TF = False
while True:
    print('*****touch A2= ' + str(touch.value) + ', sleep_acum= ' + str(sleep_acum) +', UpDn: '+ str(sleep_direction) + ', sleeping= ' + str(sleeping_TF))
#     print((sleep_acum,))

    if (touch.value == True):  # this can freeze program until I untouch???? WIP
        sleep_acum += sleep_direction

    sleep_acum = max([sleep_acum, 0])  # prevent counting down below zero.
    sleep_acum = min([sleep_acum, sleep_max])  # prevent counting above maximum zero.

    if sleep_acum > sleep_max/2:
#         pixels.fill((0, 0 , 100))  # (R,G,B) show sleep mode
#         pixels.show()
        micro_com_out.value = True
        sleeping_TF = True
        # print(' ***** sleeping ****** ' )
        
    if sleep_acum <= 0:
        sleep_direction = 1  # flip sleep increment direction
    if sleep_acum >= sleep_max:
        sleep_direction = -1  # flip sleep increment direction

    time.sleep(0.2)
```

### Behavior

I'm trying to build a sleep mode for my code using touch capacitance on A2. I've got a timer to require a long touch event on A2. I can see the "sleep_acum" counting up or down then freeze and hold the last value. The freeze event ends, code start running as soon as I remove my finger from A2.

Maybe when I'm touching A2, my finger is touching a few of the pins on CPU?

dhalbert, 2023-04-06:
I also tried most of your `while True:` loop, but removed the NeoPixel and I2C code. I then tried this bare-bones test.I could not make it hang. I was testing the A2 pin directly with the board on a breadboard.
```py
import board
import time
import touchio

touch = touchio.TouchIn(board.A2)

while True:
    print(touch.value)
    time.sleep(0.2)
```
If you remove the I2C and/or NeoPixel from your own code, does it make any difference? How is the touch wired? A picture would be helpful.

dhalbert, 2023-04-11:
I am going to close this now, since I can't reproduce. @teejay0909 Please reopen or post further if you have further information.

closed 2023-04-11 by dhalbert