adafruit/circuitpython #7770: Pico W crashes when accessing web workflow interface

RetiredWizard, 2023-03-22:
### CircuitPython version

```python
Adafruit CircuitPython 8.1.0-alpha.2-16-g60a9c7e5b on 2023-03-22; Raspberry Pi Pico W with rp2040
```

### Code/REPL

```python
No code executed, tested without a code.py
```

### Behavior

During my testing I first dismounted the CIRCUITPY drive and then attempted to access the web workflow URL. on commit  https://github.com/adafruit/circuitpython/commit/60a9c7e5b2570a84c07c6c0c78f73c63322937f0 and newer the Pi Pico W would restart as soon as I clicked on the link. If I had an serial console open an `OSError: [Errno 11] EAGAIN` message was displayed and then the board would restart. Even though the board appeared to restart once the error occurred I was unable to reconnect to the serial port until I power cycled the board.

The build from this commit: https://github.com/adafruit/circuitpython/commit/09f84e351318f9f3bf5bff6bb9664703ca743c78 did not seem to have the issue.

jepler, 2023-03-22:
#7623 introduced a problem that I believe could cause crashes in the web worfklow, but we thought we promptly fixed it in #7632. The ref you are testing at, 60a9c7e5b, is within the range of commits known to be affected. Can you check whether this problem is still affecting current versions?

RetiredWizard, 2023-03-22:
I am seeing the issue on the latest build from circuitpython.org. I am a bit concerned I'm missing something because the commit that the problem seems to start at doesn't make a lot of sense for this type of issue and I'm surprised that no one else has reported this by now. I did try plugging the pico w into both a Windows 8 PC and a Linux box and got the same result. I also connected the Pico to a usb charger so there was no computer involved and confirmed that web workflow didn't work in that case either.

jepler, 2023-03-22:
Just to eliminate doubt can you show the boot_out.txt from the "latest version" you're using?

RetiredWizard, 2023-03-22:
```
Adafruit CircuitPython 8.1.0-beta.0-48-gcfedcd411 on 2023-03-22; Raspberry Pi Pico W with rp2040
Board ID:raspberry_pi_pico_w
UID:E6614103E71F6622
MAC:28:CD:C1:00:43:35
```

DavePutz, 2023-03-22:
I am also reproducing this on the latest builds (both downloaded and generated locally). The EAGAIN error (Errno 11) is being raised in socketpool_socket_recv_into() due to the change introduced in PR #7623. However, that identical code is working fine in CP 8.0.4.
Trying to see what other CP 8.1 changes may be affecting this.

PR #7779 (2023-03-23): Revert "Correctly raise OS error in socketpool_socket_recv_into()"
This reverts commit 7e6e824d5655026906b6515070aeb604d2ef3426.

Fixes #7770

The change in #7623 needs to be revered; the raise-site added in #7632 is the correct one and the one in socketpool needs to be reverted.

This is not affecting 8.0.x because #7623 was not back-ported to there before we realized it was not a full fix.

Both #7770 and #7606 should be re-tested. I didn't test.
commit e05f0ba3 (2023-03-23): Revert "Correctly raise OS error in socketpool_socket_recv_into()"

This reverts commit 7e6e824d5655026906b6515070aeb604d2ef3426.

Fixes #7770

The change in #7623 needs to be revered; the raise-site added in #7632
is the correct one and the one in socketpool needs to be reverted.

This is not affecting 8.0.x because #7623 was not back-ported to there
before we realized it was not a full fix.

Both #7770 and #7606 should be re-tested. I didn't test.
closed 2023-03-23 by dhalbert