adafruit/circuitpython #7778: Adafruit 2.9 EPD UC8151D - Shows all black screen

VictorioBerra, 2023-03-22:
Just got this product in: https://www.adafruit.com/product/1028

Using it with this product: https://www.adafruit.com/product/5477

Exactly the same as this issue: https://github.com/adafruit/Adafruit_CircuitPython_IL0373/issues/28

CircuitPython Version: `Adafruit CircuitPython 8.1.0-beta.0-50-ge1f16472c on 2023-03-22; Adafruit Feather ESP32S3 4MB Flash 2MB PSRAM with ESP32S3`

[image]

Wiring:

| EPD  | ESP32-S3 |
| ------------- | ------------- |
|  VIN | 3V3 |
|  3V3 | |
| GND | GND |
| SCK | SCK |
| MISO | MI |
| MOSI | MO |
| ECS | D9 |
| D/C | D10 |
| SRCS | D6 |
| SDCS | D5 |
| RST | |
| BUSY | D11 |
| ENA |    |

Code (pretty much exactly from he sample, I only remove the reset pin.):

```python
# SPDX-FileCopyrightText: 2021 ladyada for Adafruit Industries
# SPDX-License-Identifier: MIT

"""Simple test script for Adafruit 2.9" 296x128 tri-color display
Supported products:
  * Adafruit 2.9" Tri-Color Display Breakout
  * https://www.adafruit.com/product/1028
"""

import time
import board
import displayio
import adafruit_uc8151d

# Used to ensure the display is free in CircuitPython
displayio.release_displays()

# Define the pins needed for display use
# This pinout is for a Feather M4 and may be different for other boards
spi = board.SPI()  # Uses SCK and MOSI
epd_cs = board.D9
epd_dc = board.D10
epd_reset = None
epd_busy = board.D11

# Create the displayio connection to the display pins
display_bus = displayio.FourWire(
    spi, command=epd_dc, chip_select=epd_cs, reset=epd_reset, baudrate=1000000
)
time.sleep(1)  # Wait a bit

# Create the display object - the third color is red (0xff0000)
display = adafruit_uc8151d.UC8151D(
    display_bus,
    width=296,
    height=128,
    rotation=270,
    busy_pin=epd_busy,
    highlight_color=0xFF0000,
)

# Create a display group for our screen objects
g = displayio.Group()

# Display a ruler graphic from the root directory of the CIRCUITPY drive
with open("/display-ruler.bmp", "rb") as f:
    pic = displayio.OnDiskBitmap(f)
    # Create a Tilegrid with the bitmap and put in the displayio group
    # CircuitPython 6 & 7 compatible
    t = displayio.TileGrid(
        pic, pixel_shader=getattr(pic, "pixel_shader", displayio.ColorConverter())
    )
    # CircuitPython 7 compatible only
    # t = displayio.TileGrid(pic, pixel_shader=pic.pixel_shader)
    g.append(t)

    # Place the display group on the screen
    display.show(g)

    # Refresh the display to have it actually show the image
    # NOTE: Do not refresh eInk displays sooner than 180 seconds
    display.refresh()
    print("refreshed")

    time.sleep(180)
```

I am using this BMP: https://raw.githubusercontent.com/adafruit/Adafruit_CircuitPython_UC8151D/main/examples/display-ruler.bmp

makermelissa, 2023-03-22:
Ok, based on that, this looks like a core issue. So I transferred it to the correct place.

tannewt, 2023-03-22:
I think I've fixed this in #7777

VictorioBerra, 2023-03-22:
How can I get a UF2 of that and test? Just wait for the nightly build to show up on S3 tomorrow?

tannewt, 2023-03-22:
Artifacts will be here once the CI completes: https://github.com/adafruit/circuitpython/actions/runs/4493575465

Once merged, it'll be in the S3 builds.

tannewt, 2023-03-23:
Ok, so the nightly with `db76` is still not working with the IL0373 but everything else is ok?

VictorioBerra, 2023-03-23:
- [ESP32-S3 TFT](https://www.adafruit.com/product/5483) - CP: `Adafruit CircuitPython 8.1.0-beta.0-58-gdb76fbd55 on 2023-03-23; Adafruit Feather ESP32-S3 TFT with ESP32S3`
    - [2.9 Color EPD UC8151D](https://www.adafruit.com/product/1028)
        - ❌ Displays gibberish
    - [2.9 Color FeatherWing IL0373](https://www.adafruit.com/product/4778)
        - ✅ Works great
- [ESP32-S3 PSRAM](https://www.adafruit.com/product/5477) - CP: `Adafruit CircuitPython 8.1.0-beta.0-58-gdb76fbd55 on 2023-03-23; Adafruit Feather ESP32S3 4MB Flash 2MB PSRAM with ESP32S3`
    - [2.9 Color EPD UC8151D](https://www.adafruit.com/product/1028)
        - ✅ Works great
    - [2.9 Color FeatherWing IL0373](https://www.adafruit.com/product/4778)
        - ✅ Works great

@tannewt

tannewt, 2023-03-24:
That is very weird. The code to drive the display should be the same so I'm not sure why it works on one but not the other. I don't think I have a UC8181D display so I'll need to order one and get back to this.

These four cases all work with 8.0.x right?

VictorioBerra, 2023-03-24:
- [ESP32-S3 TFT](https://www.adafruit.com/product/5483) - CP: `Adafruit CircuitPython 8.0.0 on 2023-02-06; Adafruit Feather ESP32-S3 TFT with ESP32S3`
    - [2.9 Color EPD UC8151D](https://www.adafruit.com/product/1028) - [Code](https://github.com/adafruit/Adafruit_CircuitPython_UC8151D/blob/main/examples/uc8151d_2.9_color.py)
        - ✅ Works great
    - [2.9 Color FeatherWing IL0373](https://www.adafruit.com/product/4778) - [Code](https://github.com/adafruit/Adafruit_CircuitPython_IL0373/blob/main/examples/il0373_2.9_color.py)
        - ✅ Works great
- [ESP32-S3 PSRAM](https://www.adafruit.com/product/5477) - CP: `Adafruit CircuitPython 8.0.0 on 2023-02-06; Adafruit Feather ESP32S3 4MB Flash 2MB PSRAM with ESP32S3`
    - [2.9 Color EPD UC8151D](https://www.adafruit.com/product/1028) - [Code](https://github.com/adafruit/Adafruit_CircuitPython_UC8151D/blob/main/examples/uc8151d_2.9_color.py)
        - ✅ Works great
    - [2.9 Color FeatherWing IL0373](https://www.adafruit.com/product/4778) - [Code](https://github.com/adafruit/Adafruit_CircuitPython_IL0373/blob/main/examples/il0373_2.9_color.py)
        - ✅ Works great

@tannewt

tannewt, 2023-04-20:
@VictorioBerra Are there any other boards that the UC8151D doesn't work with? I can't find my S3 TFT. I tried mimicking it by having my feather rp2040 EPD init a display to simulate an on board display but the UC8151D still works fine.

VictorioBerra, 2023-04-20:
@tannewt All I have is the two tri color 2.9" EPDs and the S3 TFT. I plan on getting more in the future but im just now diping my toes into all of this.

tannewt, 2023-04-20:
Ok, no problem. I'll pick up another S3 TFT with my next order.

commit 7089ea4d (2023-04-26): Add address_little_endian for displayio

Add address_little_endian for epaper displays with little endian
(low byte first) addresses.

Also clears allocated display and display bus memory so it has a
known state. The acep member wasn't always set so it varied
accidentally.

Fixes #7560. May fix #7778. Fixes #5119.
closed 2023-04-26 by dhalbert