adafruit/circuitpython #7786: [Git] Latest master cannot `make fetch-submodules`

bill88t, 2023-03-24:
On 57ac9aa763587221c2bf55dce746bd50e58159dd, which is current Master, I cannot `make fetch-submodules`.
```
fatal: remote error: upload-pack: not our ref 2b9354539e6e4f722749e87b0bdc22966dc080d9
fatal: Fetched in submodule path 'ports/mimxrt10xx/sdk', but it did not contain 2b9354539e6e4f722749e87b0bdc22966dc080d9. Direct fetching of that commit failed.
make: *** [Makefile:332: fetch-submodules] Error 128
```
Running `git submodule deinit --all -f` doesn't help.

microdev1, 2023-03-24:
Try removing it from `.git` directory `circuitpython/.git/modules/ports/mimxrt10xx/sdk` and then `make fetch-submodules`.

bill88t, 2023-03-24:
That worked, thanks, but why was that necessary. Never have I ever had to do this in the past.

closed 2023-03-24 by microdev1

dhalbert, 2023-03-24:
The old way of fetching submodules did a shallow fetch, which leaves incomplete info in the .git directory. I would like to find a clean way of doing this but haven't yet.

PR #7788 (2023-03-25): fetch-submodules fallback; add remove-submodules
Fixes #7786.

The partial clone method of fetching submodules introduced in #7763 does not work for git older than version 2.36.0. (I had hoped it was earlier than that.)

- Add a fallback to the older method, which does a shallow clone of depth 1 for each submodule.
- Add a new target `make remove-submodules`, which cleans out the currently fetch submodules, so you can start from scratch and do `make fetch-submodules` again. This is useful to switch between `fetch-submodule` methods, and also to recover from changes in submodules, such as the r [... cut]