adafruit/circuitpython #7790: PicoW: System hangs when trying to restart code after starting Wifi and webserver.

paul-1, 2023-03-25:
### CircuitPython version

```python
Adafruit CircuitPython 8.1.0-beta.0-66-gcc5451ec3-dirty

Working fine in

Adafruit CircuitPython 8.0.4-1-g4517071e6-dirty
```

### Code/REPL

```python
any of the examples from https://github.com/adafruit/Adafruit_CircuitPython_HTTPServer
```

### Behavior

Start Program,  any program interrupt after the httpserver is started will result in a hang/crash when you try to restart code.  The behavior is the same using the wifi in client or ap mode.

RetiredWizard, 2023-03-25:
I used the httpserver_cpu_information.py example program and after creating a suitable secrets.py ran the example and exited using ctrl-C. At that point if I pressed Ctrl-D the board would crash. One time I received the following message:
```
*** PANIC ***

invalid socket state for err callback
```

RetiredWizard, 2023-03-28:
The first commit in main that I see this problem on is https://github.com/adafruit/circuitpython/commit/fe40035cb9dd73d1de055ce0931c5fea99e95ffa.

dhalbert, 2023-03-28:
> The first commit in main that I see this problem on is [fe40035](https://github.com/adafruit/circuitpython/commit/fe40035cb9dd73d1de055ce0931c5fea99e95ffa).

This is the update of the CYW43 driver.

closed 2023-03-29 by dhalbert

paul-1, 2023-03-30:
Tested the latest commits, all good here.

PR #7813 (2023-03-29): don't set tcp_err callback for listen sockets (fixes #7790)
Fixes #7790.

There was a bug in the previous version of pico-sdk whereby `LWIP_ASSERT(message, condition)` would result in asserting that `message` is non-zero rather than `condition` being non-zero. This was fixed in the current version, so pre-existing assert failures are actually failing now. This is one of them.

LWIP asserts that you can't set a tcp_err callback on a listen socket, and yet we were. I think this assert is more of a sanity check rather than something that guards against invalid behavior. So I think the failure of  [... cut]
commit b863fba9 (2023-03-29): Merge pull request #7813 from gneverov/issue_7790

don't set tcp_err callback for listen sockets (fixes #7790)