adafruit/circuitpython #7795: USB_CDC DTR bad docu?

kreasteve, 2023-03-27:
### CircuitPython version

```python
8.0.4 on Pico/Tiny2040
```

### Code/REPL

```python
import usb_cdc
usb = usb_cdc.data

if usb.in_waiting:
    print(usb.readline())
```

### Behavior

No data is received if DTR is not active.

The note in https://docs.circuitpython.org/en/latest/shared-bindings/usb_cdc/index.html#usb_cdc.Serial.connected seems to relate to the whole thing, not only on `usb_cdc.Serial.connected`.

Maybe you want to update the documentation.

I tried and searched a bunch of things but I can't get it working without DTR.

### Description

I'm using hterm and c# and have this issue since the versions newer as 7.3.3

### Additional information

And related to
```
You cannot create an instance of [usb_cdc.Serial](https://docs.circuitpython.org/en/latest/shared-bindings/usb_cdc/index.html#usb_cdc.Serial). 
The available instances are in the usb_cdc.serials tuple
```
in https://docs.circuitpython.org/en/latest/shared-bindings/usb_cdc/index.html#usb_cdc.Serial

I never found an `usb_cdc.serials` tuple. I get the Serial object from `usb_cdc.data`

dhalbert, 2023-03-27:
The note in the doc says (https://docs.circuitpython.org/en/latest/shared-bindings/usb_cdc/index.html#usb_cdc.Serial.connected):

>The host is considered to be connected if it is asserting DTR (Data Terminal Ready). Most terminal programs and pyserial assert DTR when opening a serial connection. However, the C# SerialPort API does not. You must set `SerialPort.DtrEnable`.

This is an idiosyncrasy of the C# API. So if your C# code does not set  `SerialPort.DtrEnable`, `.connected` with not be true, and the behavior will be exactly you said:
>No data is received if DTR is not active.

The `tuple` reference is wrong -- that's from an older beta version of the API, sorry.

dhalbert, 2023-03-27:
If you are saying that the Note about DTR should also say that no data is delivered if DTR is not asserted, that would be a fine additin.

kreasteve, 2023-03-28:
I have a project where circuitpython is communicating with a c# program. I do not use `usb_cdc.Serial.connected` and it don't work with the normal read methods without DTR. This is confusing me, because it works until 7.3.3.

Neradoc, 2023-03-28:
The change comes from #7100 that was originally for an issue where using the serial channels too early would cause a crash, so it checks for a connection. The result is that we get a situation where there's something in the buffer, but it can't be read if nothing is connected.

I commented on that here: https://github.com/adafruit/circuitpython/issues/7456#issuecomment-1383495476
(I intended to open an issue for a discussion on that, but I wanted to do some tests first with different configurations and didn't come back to it)

dhalbert, 2026-11-13:
Claude wrote this:

This works now. On 11.0.0-alpha.3 (Feather RP2350), with `usb_cdc.data` enabled and the host opening the data port with DTR off, `usb_cdc.data.read()` received the data the host sent. Only `connected` depends on DTR, as its docstring describes.

closed 2026-11-13 by dhalbert