adafruit/circuitpython #7803: time.monotonic() is large on STM32 so seconds hidden by rounding.

calcut, 2023-03-28:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.3 on 2023-02-23; Swan R5 with STM32L4R5ZIY6
```

### Code/REPL

```python
Adafruit CircuitPython 8.0.3 on 2023-02-23; Swan R5 with STM32L4R5ZIY6
>>> import time
>>> time.monotonic()
7.33317e+08
>>> int(time.monotonic())
733317120
>>> int(time.monotonic())
733317120
>>> int(time.monotonic())
733317120
>>> time.monotonic_ns()
733317128057495129
>>> time.monotonic_ns()
733317130016052252
>>> time.monotonic_ns()
733317131590484628

Compared to ESP32S2
Adafruit CircuitPython 7.3.0 on 2022-05-23; Adafruit Feather ESP32S2 with ESP32S2
>>> import time
>>> time.monotonic()
23.922
>>> time.monotonic()
25.072
```

### Behavior

time.monotonic() doesn't give me a useful output on this STM32L4 board
I expect to be able to see seconds incrementing.
seems to be because the number is so large it is rounded when represented as a float.

time.monotonic_ns() is a workaround

Is this expected behaviour or is it avoidable?

tannewt, 2023-03-28:
This is expected. `monotonic()` is known to be limited by float precision and that's why we provide `monotonic_ns()` which uses ints that will stay accurate.

closed 2023-03-28 by tannewt

calcut, 2023-03-28:
Yep, coming from ESP32 I expected it to be initialised to zero after a hard reset, but if that is difficult for whatever reason, so be it.

Neradoc, 2023-03-29:
I did some tests and opened a new issue to follow up on the value of monotonic.  #7808 
Turns out it's 23 years because we are in 2023.