adafruit/circuitpython #7808: Setting the rtc on STM32 affects the value of time.monotonic()

Neradoc, 2023-03-29:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.4 on 2023-03-15; Adafruit Feather STM32F405 Express with STM32F405RG
```

### Code/REPL

```python
>>> import time
>>> import rtc
>>> r = rtc.RTC()
>>> time.monotonic()
30.029
>>> r.datetime = time.localtime(1680048046)
>>> time.monotonic()
7.33363e+08
>>> time.localtime(time.time() - int(time.monotonic()))
struct_time(tm_year=2000, tm_mon=1, tm_mday=1, tm_hour=0, tm_min=0, tm_sec=59, tm_wday=5, tm_yday=1, tm_isdst=-1)
>>> time.monotonic_ns()
733363727297363288
```

### Behavior

Or in other words, `time.monotonic()` returns the time between `time.time()` and 01/01/2000.
On other ports, monotonic() it is not affected by setting the RTC.

This makes monotonic not be monotonic, and lose a lot of precision in the float conversion.
`monotonic_ns` is similarly affected, but it doesn't lose precision of course.

calcut, 2023-03-29:
Thanks for looking into this Neradoc! I totally missed that I had set the RTC and not done a hard reset. That clears up some confusion.

tannewt, 2023-04-14:
I think the solution for this is to save the monotonic value when the rtc is set and the new value. Then, monotonic going forwards will be "monotonic offset + (rtc time - rtc set time)".

dhalbert, 2023-05-12:
The [CPython documentation](https://docs.python.org/3/library/time.html?highlight=monotonic#time.monotonic) says:
> Return the value (in fractional seconds) of a monotonic clock, i.e. a clock that cannot go backwards. The clock is not affected by system clock updates. The reference point of the returned value is undefined, so that only the difference between the results of two calls is valid.

PR #7965 (2023-05-12): STM: monotonic time even when RTC is changed
- Fixes #7808

Test:
```py
>> import time,rtc
>>> r = rtc.RTC()
>>> print(time.monotonic()); r.datetime = time.struct_time((2023,1,1,0,0,0,0,0,0)); print(time.monotonic())
1329.49
1329.49
>>> print(time.monotonic()); r.datetime = time.struct_time((2022,1,1,0,0,0,0,0,0)); print(time.monotonic())
1339.38
1339.38
>>> print(time.monotonic_ns()); r.datetime = time.struct_time((2022,1,1,0,0,0,0,0,0)); print(time.monotonic_ns())
1353792083741
1353791992187
>>> print(time.monotonic_ns()); r.datetime = time.struct_time((2024,1,1,0,0,0,0,0,0)); print(time.mon [... cut]
closed 2023-05-12 by tannewt