adafruit/circuitpython #7817: PulseIn does not work on FeatherS3

jschachter, 2023-03-29:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.4 on 2023-03-15; FeatherS3 with ESP32S3
```

### Code/REPL

```python
import board, pulseio

p = pulseio.PulseIn(board.D12) # physical pin 10
while True:
    if len(p) > 0:
        print(p[0])
        p.clear()
```

### Behavior

No output.

### Description

Connected a RC receiver outputting PWM (3.3 volt output, I checked w/ scope).

The same code does output a series of numbers on a Raspi Pico.

Also discussed here: https://forums.adafruit.com/viewtopic.php?p=966813

RetiredWizard, 2023-03-31:
I ran the test code using CP 8.0.5 on an ESP32-S3-DevKitC-1-N8R2 and an Adafruit Feather ESP32 S3 4/2 but didn't get any output. When I ran it on a Pico, it did display output as expected.

jschachter, 2023-03-31:
@DavePutz How are you generating the input signal? I will see if I can reproduce.

@RetiredWizard I noticed intermittent restarts on the ESP even if it wasn't detecting anything. Not sure if this is the same cause.

DavePutz, 2023-04-01:
@jschachter - I am running this script on a Pico using pulseout to generate the signals.
```
import pwmio
import pulseio
import board
from time import sleep
import array

pulse = pulseio.PulseOut(board.GP27, frequency=40000, duty_cycle=2**16 - 1)
data = array.array("H",[373, 215, 285, 327, 479])
while True:
    pulse.send(data)
    sleep(1)
```
I did also note that breaking out of the receive loop with CTRL-C sent the ESP32S3 into safe mode. I
believe that is a separate issue, so I'll take a look at what that may be causing that.

RetiredWizard, 2023-04-01:
I was simply using `pwmio.PWMOut(PinNumber,5000,5000)` to generate the test signal (which does work when received by a Pi Pico and not the S3 boards) but when I used the above script with `pulseio.PulseOut` the receiving program worked on the S3 boards as well.

HelenFoster, 2023-04-01:
Seems continuous signals are ignored on purpose on `espressif` ports (I'm not sure what the reason is).

https://github.com/adafruit/circuitpython/blob/e8711ee836eba9248cf4a73cd3e06ecf91d3e15c/ports/espressif/common-hal/pulseio/PulseIn.c#L125

RetiredWizard, 2023-04-01:
I assume this is because an 8.0.5 PR hasn't been moved to the main branch yet, but when I use 8.1.0beta.1 on the S3 boards the receiving program doesn't display any output, switching back to 8.0.5 results in the 330's being displayed again.

dhalbert, 2023-04-01:
> I assume this is because an 8.0.5 PR hasn't been moved to the main branch yet, but when I use 8.1.0beta.1 on the S3 boards the receiving program doesn't display any output, switching back to 8.0.5 results in the 330's being displayed again.

The `main` branch is up to date with fixes from the `8.0.x` branch. I think you are seeing a regression due to newer changes that are only on `main`.

RetiredWizard, 2023-04-01:
Okay, I'll keep poking at it :grin:

RetiredWizard, 2023-04-01:
I think I found the regression in this https://github.com/adafruit/circuitpython/commit/1063ec064961ab40121f1aebf2746de09a6cedcf merge to main. Specifically, If I undo the following change:
[image]
and add
```python
void port_background_tick(void) {
}
```
to the current main branch, the S3 board running the test program above displays values of 330 when the test transmit program is run and connected.

I'm still concerned that the 330 values aren't correct and the S3 should be displaying a sequence similar to the one that the Pico displays.

jschachter, 2023-04-02:
Using @DavePutz's code on a pico, and the following code on the FeatherS3 (running 8.0.4), I get the following response, which matches the first and last pulse.

```
[330, 483]
[330, 483]
[330, 483]
[330, 483]
[330, 483]
[330, 483]
```

import board, pulseio

```
p = pulseio.PulseIn(board.D12)
print(p.paused)
while True:
    if len(p) >0:
        x = []
        for i in range(len(p)):
            x.append(p[i])
        print(x)
        p.clear()
```
[image]

jschachter, 2023-04-02:
Emulating a servo pwm output, on the pico and reading the input with the code above yields no output.

```
# pico
import pwmio
import board
from time import sleep

servo_center = int(1.5 / 20.0 * (2**16))
pwm1 = pwmio.PWMOut(board.GP22, duty_cycle=servo_center, frequency=50)

while True:
    sleep(0.1)
    print('hi')
```

[image]

Code on the ESP32S3:
```
# esp32 
import board, pulseio
p = pulseio.PulseIn(board.D12)
print(p.paused)
while True:
    if len(p) >0:
        x = []
        for i in range(len(p)):
            x.append(p[i])
        print(x)
        p.clear()
```

DavePutz, 2023-04-02:
I'm not sure why, but this script shows all the incoming pulse values, not just the last 2:
```
import board, pulseio
p = pulseio.PulseIn(board.IO12,maxlen=600, idle_state=True)
print(p.paused)
while True:
    if len(p) >0:
        x = []
        for i in range(len(p)):
            x.append(p[i])
        print(x)
        p.clear()
```
The only differences are the addition of maxlen and idle_state on the PulseIn init line.

jschachter, 2023-04-02:
@DavePutz idle_state is unneeded, and maxlen defaults to 2, thus getting the two measurements.

jschachter, 2023-04-02:
I think pulseio.PulseIn cannot detect single pulses.

on pico, as above, except ```data = array.array("H",[1500, 1500, 1500])```

on esp32, as above, ```[1506, 1503, 1503]``` repeating. 

[image]

DavePutz, 2023-04-02:
Looks like the pwm not showing results is  due to this line in Espressif common-hal/pulseio/PulseIn.c:
`
   config.rx_config.idle_threshold = 30000; // 30*3=90ms idle required to register a sequence
`
Using a PWM frequency higher than 10 will result in an idle less than 90 ms; and so the pulsein read will never complete. The PWM example on the pico will result in pulses being seen on the ESP32S3 if frequency is 10 or less. I am not sure of the reason the pulsein idle_threshold was set to this seemingly arbitrary value; perhaps someone else could answer that question.

jschachter, 2023-04-03:
@DavePutz fwiw default is 12000 in https://github.com/espressif/esp-idf/blob/c9646ff0beffc86d2c6d1bfbad34da16e328e0e3/components/driver/include/driver/rmt.h#L126

jschachter, 2023-04-10:
this still does not work as expected.

closed 2023-04-10 by jschachter

jschachter, 2023-04-10:
> @jschachter <https://github.com/jschachter> did you delete your remark
> that it wasn't working?

Yes, I should not have replied.