adafruit/circuitpython #7827: metro m7 + neokey 1x4s = OSError: [Errno 5] Input/output error

jepler, 2023-04-03:
### CircuitPython version

```python
Adafruit CircuitPython 8.1.0-beta.1-17-gb011468251-dirty on 2023-04-03; Metro MIMXRT1011 with IMXRT1011DAE5A
Board ID:metro_m7_1011
```

### Code/REPL

```python
import board
from adafruit_neokey.neokey1x4 import NeoKey1x4
keys = NeoKey1x4(board.STEMMA_I2C())

while True:
    print(f"{keys.digital_read_bulk(0xff):08b}")
    time.sleep(.01)
```

### Behavior

The code runs for awhile (prints lines like "11110000" if no keys are pressed, but eventually (after way under 1 minute in my tests; faster if the sleep is removed):
```
Traceback (most recent call last):
  File "code.py", line 6, in <module>
  File "adafruit_seesaw/seesaw.py", line 216, in digital_read_bulk
  File "adafruit_seesaw/seesaw.py", line 466, in read
OSError: [Errno 5] Input/output error
```

After that, communication with the neokey is impossible until the neokey is power cycled (just resetting the m7 with the reset button does not fix it):
```
Traceback (most recent call last):
  File "code.py", line 1, in <module>
  File "code_synth.py", line 11, in <module>
RuntimeError: No pull up found on SDA or SCL; check your wiring
```

### Description

Maybe a problem with clock stretching on this mcu?

jepler, 2023-04-04:
Limor points out that the neokey 1x4 does not use clock stretching so it's not that

jepler, 2023-04-04:
however it may be specific to this device in some way. I also used a Nunchuk and it's fine so far...
```py
import board
from wiichuck.nunchuk import Nunchuk
controller = Nunchuk(board.STEMMA_I2C())

while True:
    print(controller.values)
```

dhalbert, 2023-04-07:
I am seeing the same error immediately with a BNO055, which does clock stretching.

dhalbert, 2023-04-12:
I tried this on a production (non-green) board, and could not get it to fail after about 10 minutes of waiting. I made the loop as fast as possible:
```py
import board
from adafruit_neokey.neokey1x4 import NeoKey1x4
keys = NeoKey1x4(board.STEMMA_I2C())

while True:
    keys.digital_read_bulk(0xff)
```
I am going to close this for now, but please reopen if you see it again on a production board.

closed 2023-04-12 by dhalbert