adafruit/circuitpython #7828: Aesio module not compatible with AES from python Crypto ?

ImTheBadWolf, 2023-04-03:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.3 on 2023-02-23; Raspberry Pi Pico W with rp2040
Adafruit-Blinka 8.16.1 on Raspberry pi 2B
```

### Code/REPL

```python
Pi pico:
AES_KEY = "Sixteen byte key"
message = "hello world worldd"
encrypted_message = bytearray(len(message))
cipher = aesio.AES(AES_KEY, aesio.MODE_CTR, AES_KEY)
cipher.encrypt_into(bytes(message, "utf-8"), encrypted_message)

RPI 2b:
from Crypto.Cipher import AES

AES_KEY = "Sixteen byte key"
crypto = AES.new(AES_KEY, AES.MODE_CTR, AES_KEY, counter=lambda: bytes(AES_KEY, "utf-8"))
decrypted_message = crypto.decrypt(bytes(encrypted_message))
```

### Behavior

Output on the RPI 2b, after decrypting is:  
----------------68 65 6C 6C 6F 20 77 6F 72 6C 64 20 77 6F 72 6C 17 A0 (hex)
But should be:  68 65 6C 6C 6F 20 77 6F 72 6C 64 20 77 6F 72 6C 64 64

### Description

When the length of message is larger than 16 characters, decrypted message doesn't correspond to the message which was encrypted at the beginning.

It is possible that I use the AES library on rpi 2b wrong. Mainly the counter argument, but im not sure as it is working for shorter messages.
Alternatively is it possible to use aesio module on RPI ?

jepler, 2023-04-03:
Hi! I'm not sure either.

We test our AES implementation against a small number of test cases from the publication NIST Special Publication 800-38A, 2001 edition.  The test itself is found in https://github.com/adafruit/circuitpython/blob/5909ceacc6975906ad23b07ecd011511146e07d1/tests/circuitpython/aes.py and is simply checked against an "expected output file" aes.py.exp.

the test does not cover incomplete blocks or encrypting more than 1 block in a single call. However, I started adding a test for incomplete blocks (I will show details here later today) but did not see any problem when it came to the test vectors.

Maybe there's a 3rd implementation we can compare against to find out who is right?

jepler, 2023-04-03:
Here's the added test
```py
key = unhexlify("2b7e151628aed2a6abf7158809cf4f3c")
counter = unhexlify("f0f1f2f3f4f5f6f7f8f9fafbfcfdfeff")
cyphertext = bytearray(len(plaintext))
cipher = aesio.AES(key, aesio.MODE_CTR, IV=counter)
for i in range(0, len(plaintext), 16):
    output = memoryview(cyphertext)[i : i + 16]
    cipher.encrypt_into(plaintext[i : i + 16], output)
    print(str(hexlify(output), ""))
print()

plaintext = bytearray(len(plaintext))
cipher = aesio.AES(key, aesio.MODE_CTR, IV=counter)
for i in range(0, len(plaintext), 16):
    output = memoryview(plaintext)[i : i + 16]
    cipher.decrypt_into(cyphertext[i : i + 16], output)
    print(str(hexlify(output), ""))
print()

print("truncated-CTR-2")
## Truncated CTR test case #2
plaintext = unhexlify(
    "6bc1bee22e409f96e93d7e117393172a"
    "ae"
)

key = unhexlify("2b7e151628aed2a6abf7158809cf4f3c")
counter = unhexlify("f0f1f2f3f4f5f6f7f8f9fafbfcfdfeff")
cyphertext = bytearray(len(plaintext))
cipher = aesio.AES(key, aesio.MODE_CTR, IV=counter)
cipher.encrypt_into(plaintext, cyphertext)
for i in range(0, len(plaintext), 16):
    output = memoryview(cyphertext)[i : i + 16]
    print(str(hexlify(output), ""))
print()

plaintext = bytearray(len(plaintext))
cipher = aesio.AES(key, aesio.MODE_CTR, IV=counter)
cipher.decrypt_into(cyphertext, plaintext)
for i in range(0, len(plaintext), 16):
    output = memoryview(plaintext)[i : i + 16]
    print(str(hexlify(output), ""))
print()
```

Here's the output including the original test:
```
CTR
874d6191b620e3261bef6864990db6ce
9806f66b7970fdff8617187bb9fffdff
5ae4df3edbd5d35e5b4f09020db03eab
1e031dda2fbe03d1792170a0f3009cee

6bc1bee22e409f96e93d7e117393172a
ae2d8a571e03ac9c9eb76fac45af8e51
30c81c46a35ce411e5fbc1191a0a52ef
f69f2445df4f9b17ad2b417be66c3710

truncated-CTR-1
874d6191b620e3261bef6864990db6ce
9806

6bc1bee22e409f96e93d7e117393172a
ae2d

truncated-CTR-2
874d6191b620e3261bef6864990db6ce
98

6bc1bee22e409f96e93d7e117393172a
ae
```

the first truncating test does a 16-byte block followed by a 2-byte block, while the second truncating test does 17 bytes in a single call.

due to the way CTR works, my understanding is that the result is correct, since the bytes of the truncated versions match the whole version.

FoamyGuy, 2023-04-03:
@ImTheBadWolf I do not think it's possible to use that module with Adafruit-Blinka. It would need to have a python implementation here: https://github.com/adafruit/Adafruit_Blinka/tree/main/src in order be importable and usable inside the Blinka context on a RPi.

jepler, 2023-04-03:
I tried the newer "cryptography" library (https://cryptography.io/en/latest/hazmat/primitives/symmetric-encryption/#cryptography.hazmat.primitives.ciphers.modes.CTR) and it appears to 'work' compatibly with CircuitPython.

```py
from binascii import hexlify, unhexlify
import os

from cryptography.hazmat.primitives.ciphers import Cipher, algorithms, modes

plaintext = unhexlify(
    "6bc1bee22e409f96e93d7e117393172a"
    "ae2d8a571e03ac9c9eb76fac45af8e51"
    "30c81c46a35ce411e5fbc1191a0a52ef"
    "f69f2445df4f9b17ad2b417be66c3710"
)

key = unhexlify("2b7e151628aed2a6abf7158809cf4f3c")
counter = unhexlify("f0f1f2f3f4f5f6f7f8f9fafbfcfdfeff")

def roundtrip(plaintext, key, counter):
    cipher = Cipher(algorithms.AES(key), modes.CTR(counter))
    encryptor = cipher.encryptor()
    cyphertext = encryptor.update(plaintext) + encryptor.finalize()

    for i in range(0, len(plaintext), 16):
        output = memoryview(cyphertext)[i : i + 16]
        print(str(hexlify(output), "ascii"))
    print()

    cipher = Cipher(algorithms.AES(key), modes.CTR(counter))
    decryptor = cipher.decryptor()
    plaintext = decryptor.update(cyphertext) + decryptor.finalize()

    for i in range(0, len(plaintext), 16):
        output = memoryview(plaintext)[i : i + 16]
        print(str(hexlify(output), "ascii"))
    print()

roundtrip(plaintext, key, counter)
roundtrip(plaintext[:18], key, counter)
```

If at this point you think the problem is not in circuitpython's `aesio` module, please close this issue up. Thanks!

closed 2023-04-03 by ImTheBadWolf