adafruit/circuitpython #7847: (nRF52) Use of the ARM CryptoCell for cryptography-related operations?

applecuckoo, 2023-04-08:
I was checking if `os.urandom` was available on the nRF52840 board that I use (Circuit Playground Bluefruit) and I saw that the function is unavailable on the nRF52 series. I've seen that Adafruit have an Arduino crypto library for the nRF52 series, and according to [Nordic's documentation](https://infocenter.nordicsemi.com/index.jsp?topic=%2Fps_nrf9160%2Fcryptocell.html) it seems it covers the cryptography functions that are implemented in CircuitPython, namely AES, hashing, and TRNG. Are there any specific reasons why the CryptoCell can't be used in the CircuitPython implementation?

jepler, 2023-04-09:
I thought os.urandom was expected to work on nrf52: https://github.com/adafruit/circuitpython/blob/50e259fefc0712e43ff2b894a5cb1f2aae7c9a81/ports/nrf/common-hal/os/__init__.c#L66

What are you seeing when you try to call it? I don't have any nrf boards handy to test.

applecuckoo, 2023-04-09:
Ah - my bad, seems like the documentation from https://docs.circuitpython.org/en/latest/shared-bindings/os/index.html#os.urandom is outdated, it told me that it wasn't available on nRF52 boards.

jepler, 2023-04-10:
Aha, thanks--I'm going to file a separate issue about the incorrect documentation so this issue can be specifically about the use of trust cell.

jepler, 2023-04-10:
* #6751

tannewt, 2023-04-10:
> Are there any specific reasons why the CryptoCell can't be used in the CircuitPython implementation.

Not that I know of. We just haven't added it.

applecuckoo, 2023-06-03:
Pinging @hathach for his work on the Adafruit_nRFCrypto Arduino library