adafruit/circuitpython #7871: ESP32-S2 A0 DAC truncates and reverses value below mid-point

CedarGroveStudios, 2023-04-17:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.5 on 2023-03-31; Adafruit Feather ESP32-S2 TFT with ESP32S2
```

### Code/REPL

```python
import board
from analogio import AnalogOut

a0 = AnalogOut(board.A0)
a1 = AnalogOut(board.A1)

while True:
    for i in range (0, 65535, 64):
        a0.value = i
        a1.value = i
```

### Behavior

[image]

When instantiated, the A0 DAC (top yellow trace) defaults to ~2.96v. When `a0.value` is set to 0, the voltage drops to ~1.82v instead of 0v. As the value is increased, the voltage output decreases to ~1.6v until it approaches mid-point when the output begins to track the input value as expected.

The A1 DAC output (bottom blue trace) is simultaneously tracking the same value as A0. The A1 DAC output voltage is working as expected.

anecdata, 2023-04-18:
I don't have a logic analyzer, but `AnalogIn` measurements indicate the same issue with:
`Adafruit CircuitPython 7.3.3 on 2022-08-29; Adafruit Feather ESP32S2 with ESP32S2`
It's amazing this hasn't been noticed until now.

Mini module:
> 2 IO18 on the module is pulled up to VDD33 through a 10 kΩ resistor. For details, please refer to Figure
5 and Figure 6."
(<https://www.espressif.com/sites/default/files/documentation/esp32-s2-mini-1_esp32-s2-mini-1u_datasheet_en.pdf> p.11 via <https://www.esp32.com/viewtopic.php?t=27869>)
An odd design decision by Espressif?

CedarGroveStudios, 2023-04-18:
> I don't have a logic analyzer, but `AnalogIn` measurements indicate the same issue with: `Adafruit CircuitPython 7.3.3 on 2022-08-29; Adafruit Feather ESP32S2 with ESP32S2` It's amazing this hasn't [...]

It’s interesting that they treat DAC2 (A0 for CircuitPython) with a pull-up, differently than DAC1 (A1). Most DAC outputs in the SAMD domain are buffered and have a low-value resistor to ground that would work in spite of a pull-up of of that size. Are the S2 DACs buffered?
I wonder if there’s a register setting that changes DAC output impedance, buffering, or linearity.

dhalbert, 2023-04-18:
GPIO18 is apparently used for serial bootloading as a TX line for UART1. I guess that's in the ROM. It is a strong design suggestion that this pin be tied to 3.3V to prevent it floating.
From https://www.espressif.com/sites/default/files/documentation/esp32-s2_hardware_design_guidelines_en.pdf:

> _GPIO18 works as U1RXD and is in an uncertain state when the chip is powered on, which may affect the chip’s entry into download boot mode. To solve this issue, add an external pull-up resistor._

All the ESP32-S2 modules seem to do this. We do it on the bare-chip QT Py ESP32-S2 as well, due to this recommendation. See https://learn.adafruit.com/adafruit-qt-py-esp32-s2/pinouts.

dhalbert, 2023-05-01:
I'm going to close this as "works (badly) as designed", if that's OK. I'm not sure what we can do about the pull-up that's present there. But if it's not mentioned in the guides, it should be.

CedarGroveStudios, 2023-05-01:
Certainly should be documented. Since the issue makes the pin unusable as an analog output. Shouldn’t the output capability be disabled within CircuitPython?

dhalbert, 2023-05-01:
> shouldn’t the output capability be disabled within CircuitPython?

if you added a pull-down of the same value, is the current drive enough to override the balanced pull-up/pull-down?

> Certainly should be documented.

I'll check on the guides, and we can add it to a `**Limitations:**` section in RTD.

CedarGroveStudios, 2023-05-01:
The datasheet says that the DAC output is buffered, apparently not enough to negate the effect of the pull-up resistance. That’s if the assumption of the pull-up being the cause is true — don’t know how to verify it without more design details. That’s a long-winded way of agreeing that “it’s just the way it is.”

CedarGroveStudios, 2023-05-01:
> if you added a pull-down of the same value, is the current drive enough to override the balanced pull-up/pull-down?
>
>
> I'll check on the guides, and we can add it to a `**Limitations:**` section in [...]

That’s an interesting idea. Adding an offsetting pull-down could reveal more about the buffer but would defeat the purpose of having the pull-up. I’m inclined to chalk it up to a design anomaly and leave it to the vendor to resolve.

dhalbert, 2023-05-01:
You could have a switchable pull-down by using a pin: external resistor you ground or not, or attach another input. But that's getting into the weeds.

I have no idea why they decided to add a pull-up on that pin and sacrifice the analog functionality -- it's weird.

CedarGroveStudios, 2023-05-01:
> You could have a switchable pull-down by using a pin: external resistor you ground or not, or attach another input. But that's getting into the weeds.
> 
> I have no idea why they decided to add a pull-up on that pin and sacrifice the analog functionality -- it's weird.

If we assume that the pull-up resistor is impacting the DAC, adding an external resistance or source current to the pin would introduce other non-linearity effects and temperature issues. Designers take the DAC accuracy/linearity specs seriously sometimes.

dhalbert, 2026-11-13:
Closed by #10160

closed 2026-11-13 by dhalbert