adafruit/circuitpython #7875: Web Workflow CORS implementation too restrictive

thess, 2023-04-18:
I am not sure the intent of the CORS implementation in Web Workflow is correct. From my  
limited understanding of the feature, it is not for server based request validation but
for informing the client what domains are OK for requests following. As it is currently
implemented in "web_workflow.c:_origin_ok", HTTP requests are rejected if the "Origin:"
HTTP header attribute is set and does not match one of the built-in domains or IP.
Returning a 403 error in this case does not seem correct. It should not matter what
host-name is used in the initial request.                                                           
                                                                                                    
### From the Mozilla documention of CORS
Cross-Origin Resource Sharing (CORS) is an HTTP-header based mechanism that allows
a server to indicate any origins (domain, scheme, or port) other than its own from
which a browser should permit loading resources. CORS also relies on a mechanism by
which browsers make a "preflight" request to the server hosting the cross-origin
resource, in order to check that the server will permit the actual request. In that
preflight, the browser sends headers that indicate the HTTP method and headers that
will be used in the actual request.

### Protocol description
https://fetch.spec.whatwg.org/#http-cors-protocol

[I don't think a 403 response should be returned for a request unless the Origin: is
not one that has been specified by the __client__ in the original request response.]

### Proposed change
Implement server-side CORS validation as follows:
1. If No Origin: specified, then return OK 
2. If Origin: does not have http: scheme, treat as # 1 above
3. If No Host given, and Origin: specified, return 403
4. If both Host: and Origin: are specified, check for hostname match (including port #)
   Origin: is OK if same as Host: --> return Origin:, otherwise 403
5. Special debugging case allowed for Origin: equal "localhost" (ignoring port #)

### Additional problem
Circuitpython web-editor (https://github.com/circuitpython/web-editor) has built-in
fixed host names to be allowed. This seems to be very restrictive to how host names
may exist within the network. These name restrictions are a bit arbitrary.
- Anything starting with 'cpy-' folowed by 6 hex characters and optional numeric segment.
  Ex: cpy-053ba9-2345
- Any IP address!?
- 'localhost'
- circuitpython.org will be re-directed to local mDNS name ;)

After implementing the proposed CORS change, the web-editor should have no host name
retrictions though you may still need special case for 'localhost' debugging.

tannewt, 2023-04-19:
I implemented a lot of the CORS stuff before realizing browsers wouldn't allow `.local` access from non-local pages. So, it's probably overly complex.

thess, 2023-04-19:
It was never the complexity - just the restrictions on device naming that irked me. I couldn't figure out what problem it was solving and it seemed a bit backward (client/server roles reversed) from the spec. If there is no real specific need for it, it should be removed.

thess, 2023-04-20:
I would say the device/server side is fixed if CORS support is desired (What problem is is solving??? Example, please).

The REST / Serial API is now fully functional independent of device host name used. As far as the web-editor goes, that is not fixed - it needs some code changes to allow normal naming without restrictions. 

BTW - not all environments / platforms support mDNS out-of-the-box and/or may not resolve **.local** names. mDNS does not traverse subnets. However, there exists proxies, in some routers, to forward mDNS broadcasts.

tannewt, 2023-04-21:
I think I was using CORS to call from HTML and JS hosted by the local httpserver to a circuitpython device. That way I could iterate on the html and js without rebuilding and loading CP each time.

Neradoc, 2023-04-21:
> I would say the device/server side is fixed if CORS support is desired (What problem is it solving??? Example, please).

Here is how I understand this:

The first implementations of tools using the web workflow were code.circuitpython.org trying to load API endpoints from the circuitpython.local web workflow. This would cause the browser to send a CORS preflight request to the board, so that request had to be allowed to go through, but not from any address for security reasons. I was myself doing tests with a site hosted on localhost on my computer, which was also blocked because of the preflight request. (Which I turned into an app that doesn't have that issue).

> CORS also relies on a mechanism by which browsers make a "preflight" request to the server hosting the cross-origin resource, in order to check that the server will permit the actual request.

Site A (code.circuitpython.org) tries to fetch() a DELETE request on site B (circuitpython.local) `/fs/settings.toml`.
```js
fetch("http://circuitpython.local/fs/settings.toml", {
    method: "DELETE",
    Authorization: 'Basic ' + btoa(`:${password}`),
})
```
Do we allow it on the server side ?
So the solution was to allow cross-site requests from code.circuitpython.org with the CORS response.
But disallow: `malicious.evil` trying to DELETE `http://circuitpython.local/fs/settings.toml`.

I believe this is normally used when `www.website.tld` wants to make calls to `api.website.tld` or `static.website.tld`, which will then respond OK to the preflight only if the request's origin is part of the same domain for example (it is a cross site request because it's not the same subdomain).

An example of that is also my first attempt at [making a library installer in a web page](https://www.neradoc.me/bundler/), as github blocks cross site access to files, so a web page couldn't dynamically load the bundle zip, making me create a proxy script. The proxy downloads the file (it can because, not being a browser, it doesn't hate itself) and serves it since it's on the same site as the page (it can also respond positively to the preflight request).

We could allow all cross site requests, but if CORS exists it's because a malicious site could do requests to delete all your files on circuitpython.local, and those requests could go through without it, maybe if you saved the password when prompted for it (so no popup would give away the maneuver). Or actually they wouldn't anymore, but for a different reason that wasn't an issue at the time.

Now, because we can't have nice things, more recent versions of browsers block even accessing a .local url or a non-https url from an online site in some way or other, so code.circuitpython.org no longer tries to access the board's web workflow API, but instead it is loaded into the .local page, making allowing cross site requests possibly moot. (A third party web site wanting to make a tool for CP would have the same issue).

wscullen, 2023-04-24:
Just running into this where my custom local domain on my local network doesn't work properly, the OPTIONS request is 403'd. I assume it's the CORS restrictions. Would putting additional allowed CORS domains in the settings.toml be something that could be implemented at some point?

Neradoc, 2023-04-24:
Have you tested with the absolute newest ?
#7876 fixes the code limiting all access to the internally known names.

The `OPTIONS` request here is likely to be the one made by the web workflow page to get the list of allowed actions, to know if the drive is writeable (it tests the presence of DELETE in `Access-Control-Allow-Methods`).

https://github.com/adafruit/circuitpython/blob/adfb9b196bfa328b183df2412cf9bde7f9e397db/supervisor/shared/web_workflow/static/directory.js#L46-L53

Note that this is not technically a CORS issue since it's not a cross-site request with pre-flight, it's Circuitpython doing some additional filtering on the requesting websites prior to the fix.

If using latest doesn't work, can you give more details on your setup, maybe the browser log from the failing request or something ?

wscullen, 2023-04-25:
Just tried with version 8.1.0-beta.1-48-gabb0d5e62 and it does work with a custom domain. Thanks for the tip!

tannewt, 2023-11-28:
I'm not sure what is left to do. So, I'm closing this.

closed 2023-11-28 by tannewt

thess, 2023-11-28:
Agree -- Remaining issue is up-steam server side which I was able to work around.