adafruit/circuitpython #7883: Fix `abs()` warning in `vectorio/Circle.c`

dhalbert, 2023-04-20:
Tagging @kvc0 and maybe @FoamyGuy:

Compilation generates this mild warning:
```
../../shared-module/vectorio/Circle.c: In function 'common_hal_vectorio_circle_get_pixel':
../../shared-module/vectorio/Circle.c:26:22: warning: taking the absolute value of unsigned type 'uint16_t' {aka 'short unsigned int'} has no effect [-Wabsolute-value]
   26 |     int16_t radius = abs(self->radius);
      |                      ^~~
```
The `radius` struct element is already unsigned:
https://github.com/adafruit/circuitpython/blob/5014ddb43c28f7a5b7ac38ef9bf65c65d6006597/shared-module/vectorio/Circle.h#L12
I am not quite sure what is going on here with the `x` and `y` here, because they are going to be `>0` after lines 27 and 28. Maybe the original arguments should have been preserved?
https://github.com/adafruit/circuitpython/blob/5014ddb43c28f7a5b7ac38ef9bf65c65d6006597/shared-module/vectorio/Circle.c#L24-L40

kvc0, 2023-04-21:
That abs on radius should definitely be removed. Thanks for noticing!

x&y math should have abs(). Circles are placed by their center point, so the math only needs to be concerned with the 1st quadrant. The other 3 are symmetrically mirrored with positive math. You _could_ factor out the abs by doing signed math up to the end, but I think you might have to store 1 more variable if you did so. Dubious that it would read any better :-)

FoamyGuy, 2023-04-21:
I'll make this change and submit a PR for it later on this evening.

dhalbert, 2023-04-21:
@kvco I don't understand the point of the `(int32_t)x * x`, then. It does not recover the sign. What is the purpose of the `(int32_t)` casting?

kvc0, 2023-04-21:
@dhalbert widening. So that circles at 256 < x or 256 < y can still be handled. I have some Adafruit displays with more pixels than that ;-)

If you multiply 257 * 257 int16's what do you get? :-) On the other hand, if you widen to int32, you can support nearly every positive position of int16 x and y. Ideally we'd widen to uint32 instead though because (2^15)^2 * 2 is 2^31, and that's 1 greater than 2^31 - 1! But it fits no sweat in 2^32. Int16.minval x and y will position at 0 I think. So vectorio would draw 1 phantom pixel somewhere (dependent on transforms) in that unique case.

jepler, 2023-04-21:
Being explicit may be helpful but it's also the case (as far as I know) that the arithmetic will actually happen according to C promotion rules in "int" type, which is always 32 bits for us. On an 8bit micro it could be different but we're on 32 bits only.

kvc0, 2023-04-21:
So I personally prefer to be explicit when bit width matters... I always forget what the rules are for implicit widening within a math expression :-( and whether I'm on an 8, 16, 32 or 64 bit pointer machine :-(, so it helps me read it. Maybe it's helpful to other people reading it to be explicit also? It's otherwise implicitly dependent on the size of `int` (or ptr, again I'm not sure, and being explicit makes me sure).

Also, I notice that below even after abs() we're using int16_t. On arm gcc that results in using `ldrsh` while if we abs()'d to `uint16_t` we'd do `ldrh` in the math which is a simpler 0-extending widening. I don't know whether the performance is better between [a] storing 2 new uint16_t values on the stack _or_ [b] 4 instances of `ldrsh` instead of `ldrh`. I haven't checked the pixel fill rate on circle to compare the 2 options, in case you're feeling spicy :-p.

PR #7890 (2023-04-24): remove abs() from radius
Attempt to resolve #7883 

This removes the unnecessary absolute value call from radius variable. 

Honestly I don't quite follow the rest of the conversation about x and y calculations and casting. If there are other proposed changes that need to be tested out I'm happy to try those if someone can point me more specifically to what the code would look like.
closed 2023-04-24 by dhalbert