adafruit/circuitpython #7885: Using plain terminalio with displayio - ValueError: Tile index out of bounds

dorinclisu, 2023-04-21:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.5 on 2023-03-31; Raspberry Pi Pico with rp2040
```

### Code/REPL

```python
import displayio
import board
import busio
import terminalio

from adafruit_st7735r import ST7735R

displayio.release_displays()
tft_spi = busio.SPI(clock=board.GP18, MOSI=board.GP19)
tft_cs = board.GP17
tft_dc = board.GP16
tft_reset = board.GP20

displayio.release_displays()
tft_bus = displayio.FourWire(tft_spi, command=tft_dc, chip_select=tft_cs, reset=tft_reset)
tft = ST7735R(tft_bus, width=128, height=128, colstart=2, rowstart=3)

background = displayio.Group()
tft.show(background)
bitmap = displayio.Bitmap(128, 128, 1)
palette = displayio.Palette(1)
palette[0] = 0x7f007f
sprite = displayio.TileGrid(bitmap, pixel_shader=palette, x=0, y=0)
background.append(sprite)

terminal = terminalio.Terminal(sprite, font=terminalio.FONT)
terminal.write('abc')
```

### Behavior

```python
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: Tile index out of bounds
```

### Description

I'm not sure if I'm using terminalio correctly, but I found no examples and this is all the documentation: https://docs.circuitpython.org/en/latest/shared-bindings/terminalio/index.html

I know there is an extra library `adafruit_display_text` (works with the exact same setup) but plain terminalio is also supposed to work which is why I think this is a bug.

anecdata, 2023-04-21:
I'm able to do it this way:
```py
import sys
import board
import displayio
import terminalio

display = board.DISPLAY  # or equivalent external display library

splash = displayio.Group()

fontx, fonty = terminalio.FONT.get_bounding_box()
term_palette = displayio.Palette(2)
term_palette[0] = 0x000000
term_palette[1] = 0xffffff
logbox = displayio.TileGrid(terminalio.FONT.bitmap,
                            x=0,
                            y=0,
                            width=display.width // fontx,
                            height=display.height // fonty,
                            tile_width=fontx,
                            tile_height=fonty,
                            pixel_shader=term_palette)
splash.append(logbox)
logterm = terminalio.Terminal(logbox, terminalio.FONT)

display.show(splash)

print("Hello Serial!", file=sys.stdout)  # serial console
print("\r\nHello displayio!", file=logterm, end="")  # displayio
logterm.write(b'\r\nabc')
```

tannewt, 2023-04-21:
Terminal should verify that the tilegrid is more than 1x1 and raise an exception if not.

PR #7892 (2023-04-24): add minimum validation for tilegrid size in terminalio.Terminal
Attempts to resolve #7885

This PR adds minimum size validation that enforces the TileGrid being no smaller than 2x2, which is arguably still too small to be practically useful. Perhaps a higher limit would be appropriate?

If there is a desire to allow 1xN sized TileGrids the logic in this will have to get reworked.

Also something for consideration would be adding an example to the docs of terminalio module that illustrate how to use it. As noted in the issue there isn't currently any sample code for it anywhere that I'm aware o [... cut]
closed 2023-04-24 by dhalbert