adafruit/circuitpython #7888: `esp32` / `esp32c3` Serial object.

bill88t, 2023-04-21:
Currently on boards without usb_cdc it's impossible to get a `Serial` object for the main uart/serial peripheral interface.

It's needed to read `in_waiting`, `reset_input_buffer`, `reset_output_buffer` and have access to raw `read`/`write`.

tannewt, 2023-04-24:
For the ESP32 I think a board.SOMETHING for the UART object should be sufficient.

On C3 we will need to make an object for the USB to Serial/JTAG interface.

bill88t, 2023-05-13:
Perhaps, due to how few use `console` objects, it would be wise to just make a python module, to wrap the old serial functions.

It would only need to:
 - Wrap the regular serial i/o.
 - While serial bytes available is True, read them in a temporary buffer and return the count in in_waiting.

`connected` and `out_waiting` do not have to be implemented/wrapped, since they are not applicable.

I will make it and name it `virtUART`. Perhaps as a 9.0 api change, it should be considered doubling down on the nice `console` objects, replacing `runtime.serial*` stuff and instead provide a `runtime.serial` `console` object.

berkut0, 2024-08-14:
I recently discovered that I can't communicate with the ESP32-C3 based controller via serial port bypassing REPL. It's not the most convenient thing in the world! I think I'll have to swap out the ESP32-C3 for another board.