adafruit/circuitpython #7894: RP2040 DVI Zooming bitmap

jposada202020, 2023-04-22:
### CircuitPython version

```python
Adafruit CircuitPython 8.1.0-beta.1-36-g7d02bff6b on 2023-04-20; Adafruit Feather RP2040 DVI with rp2040
```

### Code/REPL

```python
import board
import displayio
from bitmaptools import draw_line
display = board.DISPLAY
group = displayio.Group()

palette = displayio.Palette(1)
palette[0] = 0xFFFFFF

a2 = displayio.Group(x=0, y=0, scale=1)
a2bitmap = displayio.Bitmap(300, 300, 2)
tile2 = displayio.TileGrid(a2bitmap, pixel_shader=palette, x=0, y=0)
draw_line(a2bitmap, 1, 1, 1, 300, 1)
draw_line(a2bitmap, 1, 1, 200, 200, 1,)
a2.append(tile2)
group.append(a2)

display.show(group)
```

### Behavior

[image]

### Description

So Not sure if my display resolution is all wrong, but I should get 640x480? If it is my mistake, sorry :).
However this script works fine in the Pyportal Titano https://github.com/jposada202020/CircuitPython_uplot/blob/main/examples/uplot_readme_example.py
But with the DVI is blown-up and some of the plots could not be seen.

dglaude, 2023-04-22:
Just sharing what this code give on my SONY Bravia TV. What was your question?
[image]

dglaude, 2023-04-22:
First thank you for your piece of code, I could not have progress without reading your code.

I changed a few things to let you check your screen rendering.
First I make the bitmap exactly the size of the DISPLAY.
`a2bitmap = displayio.Bitmap(display.width, display.height, 2)`

Then I draw a boarder one pixel inside.
Finally I use a sleep loop to stay in the program and not exit in the REPL.

`import time

import board
import displayio
from bitmaptools import draw_line
display = board.DISPLAY

group = displayio.Group()

palette = displayio.Palette(1)
palette[0] = 0xFFFFFF

a2 = displayio.Group(x=0, y=0, scale=1)
a2bitmap = displayio.Bitmap(display.width, display.height, 2)

tile2 = displayio.TileGrid(a2bitmap, pixel_shader=palette, x=0, y=0)

draw_line(a2bitmap, 1, 1, 1, display.height-2, 1)
draw_line(a2bitmap, 1, 1, display.width-2, 1, 1)
draw_line(a2bitmap, display.width-2, 1, display.width-2, display.height-2, 1)
draw_line(a2bitmap, 1, display.height-2, display.width-2, display.height-2, 1)

draw_line(a2bitmap, 1, 1, display.height-2, display.height-2, 1)

a2.append(tile2)
group.append(a2)

display.show(group)

while True:
    time.sleep(1)`

dglaude, 2023-04-22:
And the result:
[image]

dglaude, 2023-04-22:
I know your mistake, you expect 640x480... and that the output resolution after pixel doubling, you only have 320x240 in the frame buffer.
I know it from listening to "Ask an Engineer" or "Show and Tell", I am not sure where this is documented in writing.

jposada202020, 2023-04-22:
Thank you. Thats explain a lot. Happily will close this 😇. Will test later. Thank you for the info @dglaude . I guess I will need to learn more about this. 🥳

closed 2023-04-22 by jposada202020

jposada202020, 2023-04-23:
@dglaude Thank you
[image]