adafruit/circuitpython #7898: Cannot use adafruit_framebuf to draw on picodvi.Framebuffer (multiple issues)

dglaude, 2023-04-22:
### CircuitPython version

```python
8.1.0-beta.1-39-gd078bc3ae
```

### Code/REPL

```python
#Trying to draw in the framebuffer using adafruit_framebuf

import displayio
displayio.release_displays()

import time
import board
import picodvi
import framebufferio

w=640
h=480

fb = picodvi.Framebuffer(
    width=w, height=h, color_depth=1,
    clk_dp=board.CKP, clk_dn=board.CKN,
    red_dp=board.D0P, red_dn=board.D0N,
    green_dp=board.D1P, green_dn=board.D1N,
    blue_dp=board.D2P, blue_dn=board.D2N)

import adafruit_framebuf
afb=adafruit_framebuf.FrameBuffer(fb,w,h, adafruit_framebuf.MVLSB)

display = framebufferio.FramebufferDisplay(fb) # Goes to safe mode if using (afb) ok with (fb)

afb.circle(center_x=100,center_y=100,radius=90,color=1)

#]0;🐍code.py | 8.1.0-beta.1-39-gd078bc3ae\Traceback (most recent call last):
#  File "code.py", line 26, in <module>
#  File "adafruit_framebuf.py", line 378, in circle
#  File "adafruit_framebuf.py", line 358, in pixel
#  File "adafruit_framebuf.py", line 147, in set_pixel
#TypeError: 'Framebuffer' object isn't subscriptable

while True:
    time.sleep(1)
```

### Behavior

The expected behavior, or objective is to be able to draw directly in the picodvi framebuffer and that this result in displaying on the DVI connected screen.

I totally don't know what I am doing, or if I am doing that right, but I try to put together what I think is the normal or CP way of doing this... without using displayio.

So I try to use adafruit_framebuf and the drawing function available there, but working directly on the display framebuffer.

### Description

3 problems:

Problem 1: picodvi.Framebuffer vs adafruit_framebuf.FrameBuffer

picodvi does not use an upper case B for FrameBuffer, and that is hurting since previous usage have established the use of FrameBuffer ...

Problem 2: Feather RP2040 DVI goes in safe mode

if line 24 is written: `display = framebufferio.FramebufferDisplay(afb)`
Where afb is an adafruit_framebuf.FrameBuffer rather than the expected picodvi.Framebuffer.

Problem 3: a Framebuffer created by picodvi.Framebuffer is not subscritable

This is breaking all of my hopes to using existing function from adafruit_framebuf to draw something on the DVI screen.

### Additional information

I might totally misunderstand the various framebuffer and how to use those... so maybe it is not a bug but:
A) a misunderstanding... and I should discuss that on discord or in the weed
B) a feature request... because it is desirable and would save memory for some usecase
C) totally possible another efficient way... and I would like to learn

tannewt, 2023-04-24:
1. I don't think this is an issue because lower case `b` is consistent with existing native frame buffers.
2. Definitely a bug. It shouldn't crash.
3. I think it makes sense that you would be able to write directly to the FB. This is a longer term thing though because we'll want to apply it to other native frame buffers.

Rybec, 2026-09-24:
Questions:

Can this not be applied to picodvi framebuffer without applying to all of the others you want to apply it to at the same time?  Is there a reason this one has to wait so long merely because others would benefit from this feature as well?

I have a little bit of experience with making Python modules in C.  It doesn't seem like making the picodvi framebuffer subscriptable should be very difficult.  Is there some reason just making it subscriptable isn't an option?  Looking through the Python C API, it looks like this is maybe a few hour fix at the most, especially for someone more experienced in writing Python modules in C than I am.

There's a possibility I'll have some time within the next few weeks or months that I could put into making the picodvi framebuffer subscriptable, but the last response makes it sound like this is a huge undertaking because maybe it involves major decisions and adding the same capability to a bunch of other native framebuffers all in the same release cycle.  If it really is some huge thing, there's little point in me getting involved, because whatever I choose to do would likely just be thrown out, maybe because implementation details of other framebuffers make my solution impractical or impossible for them, or maybe just because someone internal decides they don't like it as much as some other solution.

---

CP _really_ needs some kind of low level graphics API.  displayio is high level almost to an extreme, in the context of low resource microcontrollers.  Having to encapsulate nearly everything, sometimes multiple layers deep, makes the code harder to read and also often to write, when you aren't trying make a high level GUI.  It seems like most of the parts exist, but the ones that aren't deprecated or abandoned are missing basic interoperability features, like subscriptable framebuffers.  I understand that many people do prefer to make GUIs that displayio is more suited to, but as someone who is involved in teaching children, I can tell you that displayio causes a lot of confusion that a lower level graphics API would avoid.  And now that we have Turbo providing the ability to compile graphics generation functions to native, this is more valuable than ever.  Access to the picodvi (or any other) framebuffer at a level that would work with native or viper would be a game changer in terms of high performance (for a microcontroller) graphics in CircuitPython.

Anyhow, I see that this has been sitting for over 3 years, with apparently no attention, even though the utility of it has been acknowledged.  Maybe now that Turbo has increased the value of this so dramatically, it will get more attention?  And if not, please at least answer my questions, and maybe explain what I would need to do to ensure that Viper can access the framebuffer correctly.  I can't guarantee I'll have time, but if this isn't valuable enough for someone at Adafruit to do, maybe I can figure it out.  And maybe even if I don't have time, someone else who does could use this information to do it.  (Note that, at least at this time, I definitely won't have time to do it for more than just picodvi.  But I might have time for that, at least.)

lynt-smitka, 2026-09-24:
@Rybec I think `picogame` might cover a lot of what you're looking for. It's a native C graphics module that landed in CircuitPython 10.3, much lower level than displayio. It draws to picodvi displays, and a canvas can use your own bytearray, so you can also write pixels from Python or viper. It's in the Fruit Jam firmware; other boards need a build with CIRCUITPY_PICOGAME=1. More details:  https://picogame.makerclass.cz/

Rybec, 2026-09-24:
> It's in the Fruit Jam firmware;

As in, it's built into the official CircuitPython build for the Fruit Jam?

I'll take a look!  In my experience game libraries tend to have very good low level graphics support, since even on desktop the overhead of a high level graphics API can be a serious problem for graphics more complex than a standard GUI.

lynt-smitka, 2026-09-24:
Yes, it's officially part of Fruit Jam from 10.3, and also on a few other boards. I hope, that number will continue to grow :-)

This should do exactly what you need:  https://picogame.makerclass.cz/concepts/engine-only/#drawing-straight-onto-a-dvi-screen

If you use LLM, you can point it here for a quick start: https://picogame.makerclass.cz/ai-agents/

Rybec, 2026-09-24:
Ok, that's pretty cool that it's part of the official build for the Fruit Jam!  That's a strong point in favor.  In fact, it will likely end up being the deciding point for me, because...

I've looked over the site, and the default and encouraged graphics pipeline is _way worse_ than displayio in terms of high level complexity.  I did end up finding the section you referenced on drawing straight to the DVI screen, but boy is it buried!  That should work for me, and since picogame is already built in, I'm not losing anything by using it.

The documentation seems rather a mess.  Finding the straight to DVI drawing myself (before you posted the link) was quite unintuitive.  I'm struggling to figure out how to do basic things like rendering an image to the screen from a file.  I've come across a few things with instructions for converting PNGs into other formats, but as far as I can tell, this is scattered throughout the documentation, and I haven't found anything on how to actually render it, aside from vague mentions about converting them into sprites without any examples of how this is done.  I'm having similar problems with figuring out how to render text.  There are at least instructions for creating labels, but I just want to render text directly to a buffer (a Canvas, I suppose, since that's the interface to the framebuffer in this instance), without using high level objects.  Is this even possible?  I'm also not seeing anything mentioning pcf font support, which is a concern.  It looks like I might be able to work around this with fontio, but it isn't clear.  This is a big deal, because bdf fonts can be easily twice the size of the same font turned into a pcf.  (I currently have a Macropad project that I made a custom font for, and it's about 4MB as a bdf and only 2MB as a pcf.  Even in the Fruit Jam, with 16MB of flash, 2MB is a big deal!)

Anyhow, right now this is so close, but if I can't render images and text directly to the canvas, that's a major roadblock.  Does picogame have any kind of simple image buffer object, that images can be loaded to and text can be rendered to?  If it even had that, I could probably write a native or viper function for blitting those to a Canvas.

lynt-smitka, 2026-09-24:
Thanks for the feedback. You are right, the docs is primarily intended for use with its game libraries and focuses on games. Raw usage is only mentioned marginally. But anything is possible 🙂

`Canvas` draws into any buffer you give it, including the `picodvi.Framebuffer` itself, and `pg.Bitmap` wraps any buffer without copying it - bytes from a file, a bytearray, a displayio Bitmap, even the DVI framebuffer - as long as the pixels are RGB565 or 8-bit palette. So you can keep using adafruit_imageload, adafruit_bitmap_font and terminalio.FONT, and picogame just does the drawing.

```
import array
import board
import displayio
import picodvi
import terminalio
import adafruit_imageload
import picogame as pg

# Draw straight into the DVI framebuffer
displayio.release_displays()
framebuffer = picodvi.Framebuffer(320, 240, color_depth=16,
    clk_dp=board.CKP, clk_dn=board.CKN,
    red_dp=board.D0P, red_dn=board.D0N,
    green_dp=board.D1P, green_dn=board.D1N,
    blue_dp=board.D2P, blue_dn=board.D2N)
screen = pg.Canvas(320, 240, buffer=framebuffer)

def color(r, g, b):
    """RGB565 in the byte order picodvi uses."""
    return ((r & 0xF8) << 8) | ((g & 0xFC) << 3) | (b >> 3)

def load_image(path):
    """Load a BMP as a picogame Bitmap"""
    bitmap, palette = adafruit_imageload.load(path, bitmap=lambda w, h, n: displayio.Bitmap(w, h, 256), palette=displayio.Palette)
    colors = array.array("H", [0] * 256)
    for i in range(len(palette)):
        rgb = palette[i]
        colors[i] = color(rgb >> 16, (rgb >> 8) & 0xFF, rgb & 0xFF)
    pixels = memoryview(bitmap)
    return pg.Bitmap(pixels, bitmap.width, bitmap.height, format=pg.PAL8,  palette=colors, stride=len(pixels) // bitmap.height)

screen.clear(color(0, 0, 40))
screen.blit(load_image("/picture.bmp"), 10, 10)
screen.text(10, 120, "Hello", color(255, 255, 255), terminalio.FONT)
```

`load_image` asks `adafruit_imageload` for an 8-bit displayio Bitmap, and picogame then reads those same pixels directly; only the palette gets converted.

For the image buffer you asked about - Canvas works off-screen too. Give it a bytearray, draw images and text into it, then blit it anywhere with pg.Bitmap(that_bytearray, w, h). Since it's a plain bytearray, your own viper code can work on the same pixels.

`Canvas.text` only works with terminalio.FONT. For a PCF font, adafruit_bitmap_font loads the glyphs, and each one becomes a small bitmap you blit:

```
from adafruit_bitmap_font import bitmap_font

font = bitmap_font.load_font("/myfont.pcf")
glyph_cache = {}

def glyph_bitmap(glyph, text_color):
    """Turn a PCF glyph into a picogame Bitmap with a transparent background."""
    width, height = glyph.bitmap.width, glyph.bitmap.height
    pixels = bytearray(width * height)
    for y in range(height):
        for x in range(width):
            pixels[y * width + x] = glyph.bitmap[x, y]
    palette = array.array("H", (0, text_color))
    return pg.Bitmap(pixels, width, height, format=pg.PAL8,
                     palette=palette, transparent=0)

def draw_text(canvas, x, y, text, text_color):
    font.load_glyphs(text)
    for char in text:
        glyph = font.get_glyph(ord(char))
        if glyph is None:
            continue
        if glyph.bitmap.width and glyph.bitmap.height:
            key = (char, text_color)
            if key not in glyph_cache:
                glyph_cache[key] = glyph_bitmap(glyph, text_color)
            canvas.blit(glyph_cache[key], x + glyph.dx, y - glyph.height - glyph.dy)
        x += glyph.shift_x

draw_text(screen, 10, 160, "PCF text", color(255, 220, 0))
```

The first time a character is drawn it gets converted, after that it's cached (~10 times faster for repeated use). 

The same Canvas works off-screen too - give it a bytearray, draw into it, and blit it with pg.Bitmap(that_bytearray, w, h).

One small complication - picodvi uses native RGB565 byte order, while pg.rgb565() gives the byte order SPI displays use. It is fixed color() function in the example.

Rybec, 2026-09-24:
I think I can work with that.  I'll have to write up some utility functions, but that shouldn't be that hard!

(Incidentally, I have some desktop game dev experience, and my typical workflow starts with raw graphics access, and then I build whatever rendering system I need.  As nice as it can be to have sprite handling built in, generic systems are rarely as good as systems custom built for the specific game.  This isn't because they are made poorly, but rather just because they have to handle everything, but most games will only use some subset of that.  Also, many game types need something other than sprite based graphics.  In my experience, most game libraries put primitive graphics access first and then treat sprite handling as a secondary module on top of that, if they include it at all.  (See  SDL2 and Pygame, for examples of this.  I think Pygame has some sprite handling, but it's not the default.  It's more of there if you need it.)  So libraries that focus on games in general rarely start with sprite handling out of the gate, and instead just provide primitive graphics access with image loading, text rendering, and blitting functions.  The few I've seen that focus heavily on sprite graphics and not much on primitive graphics are almost always optimized for a specific range of game types and ignore the fact that anything else exists.  This isn't intended as a criticism of libraries like this, but it looks like picogame is designed more to be a framework designed almost exclusively for sprite based games rather than being a full fledged game library.  This is great, if your game conforms exactly to the specific sprite handling system implemented, but otherwise it can make things very difficult.  All of that said, it sounds like it has just enough primitive graphics handling to function as a more general game dev library, if you can figure out how to use the primitive graphics stuff.)

Anyhow, thanks!  This is really helpful.  Maybe once I have more experience I'll write up my own tutorial on using picogame for lower level, more general game programming, so that in the future people like me will have a better resource.  Given that it has become an official part of CircuitPython, even if only for a handful of boards so far, that might be useful for a lot of people.